MagicMirrorOrg/MagicMirror #1249: No access to WebServer

nico0709, 2018-04-03:
Hi,
I'm using MagicMirror and it works perfectly, except the web server. When I logon from localhost:1984, I can access to the mirror interface, but not on any other device on my network... could you help me?

MichMich, 2018-04-03:
Check the documentation. You need to give a specific remote IP access.

nico0709, 2018-04-03:
Thank you for your answer. I check the file "config", and I autorize the IP adress of my computer, but it doesn't work... The documentation doesn't help me for this specific case...
I tried to reset Raspbian, but it doesn't work too... It was working on a previous version of MagicMirror²...

Hurg84, 2018-04-05:
Hi Nico,

changing 

address: "localhost",

to

address: "",

in /home/pi/MagicMirror/config/config.js should do the trick.
I'm not sure if a reboot is required.

Regards

Hurg

nico0709, 2018-04-05:
@Hurg84 Here is the part of config file with network:

var config = {
	port: 1984,
	ipWhitelist: [], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

I can access to MagicMirror online from the raspberry pi, whit Chromium, but when I use another PC it doesn't work...

Thanks,

Nico

Hurg84, 2018-04-05:
Hi Nico,

this is my config on port 8080 

Ipwhitelisting isnt important if the webserver is only listening to "localhost"

try ' address: "", ' or something similar to make the whitelisting "important" to the webserver.

var config = {
        address: "", // Address to listen on, can be:
                              // - "localhost", "127.0.0.1", "::1" to listen on$
                              // - another specific IPv4/6 to listen on a speci$
                              // - "", "0.0.0.0", "::" to listen on any interfa$
                              // Default, when address config is left out, is "$
        port: 8080,
        ipWhitelist: [], // Set [] to allow all IP addresses

MichMich, 2018-05-01:
Did you manage to solve this?

nico0709, 2018-05-04:
No, sadly...

nico0709, 2018-05-14:
@MichMich Do you have a solution please?

nico0709, 2018-06-05:
Ok, I found the soluce, it's was a problem with my Raspberry Pi

closed 2018-06-05 by nico0709

mirror-master, 2018-09-25:
MichMich has been hinting, the config file.  I'll break down how i reach mine outbound with modifying the config and using port fowarding for those confused.
Next: In your config file (at the top):  where it says local host, change that to your RPI's IP with " " 
then change your whitelist to []
last port foward on your router 
after that, use your public and port (example) 64.100.112.223:8080