MagicMirrorOrg/MagicMirror #1251: feelsLike is not translated

BerndKohl, 2018-04-06:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 2

**Node Version**: irrelevant

**MagicMirror Version**: 2.3.1

**Description**: New feature feelsLike in current weather module is not translated, always shows "Feels" indepdendant of the configured language

**Steps to Reproduce**: Set language to non-english and enable feelsLike for current weather

**Expected Results**: Translated string for "feels"

**Actual Results**: Always "Feels" is displayed.

MichMich, 2018-04-06:
Feel free to send a PR with the right translations.

BerndKohl, 2018-04-06:
If there is some kind of documentation on how to do this I would be willing to do so.

MichMich, 2018-04-06:
Maybe this helps? https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github

MichMich, 2018-04-06:
Merged: https://github.com/MichMich/MagicMirror/pull/1252

closed 2018-04-06 by MichMich