MagicMirrorOrg/MagicMirror #1254: Current weather no longer showing

sefeing, 2018-04-06:
Hi! I just updated my mirror using the update provided in the readme, and now the current weather module fails to load at all - I've tried moving to different sectors of the screen, but doesn't seem to want to display in any region.

MichMich, 2018-04-06:
Thanks for opening an issue!

To allow us to help you it's extremely important that we have all the details about your setup. During the creation of this issue, important details were asked to fill in. Because you did not provide (any) of these details we are unable to help you. 

If you are running into errors, please provide as many detailed error logs as possible in addition to the requested details.

If you are unaware how to retrieve all the requested information, feel free to ask for help on http://forum.magicmirror.builders.

This issue will be closed. Feel free to open a new issue with all the requested information.Please fill in all the request

closed 2018-04-06 by MichMich

TimothyBramlett, 2018-04-07:
@sefeing, I am having the exact same issue and I have made no changes to my config or anything.  Did you resolve this?

TimothyBramlett, 2018-04-07:
@MichMich @sefeing 

Oh I think I see the problem:

Line 415 of currentweather.js:

```
case "metric": this.feelsLike = windChillInC.toFixed(0);
```

Should be:

```
case "metric": this.feelsLike = windchillinF.toFixed(0);
```

TimothyBramlett, 2018-04-07:
@MichMich I updated my codebase with the proper variable names and verified that this fix corrected the issue with Current Weather showing up.  I then created a pull request that you can find referenced here:  https://github.com/MichMich/MagicMirror/pull/1256