MagicMirrorOrg/MagicMirror #1263: Heat index "feels like" reporting inaccurate numbers

parnic, 2018-04-18:
**Platform**: Windows 10 (run from WSL Bash) and viewed with Chrome 65. Also seen with RPi Zero W host with Midori 0.4.3 browser.

**Node Version**: Seen with LTS 8.10.0 and 8.11.1

**MagicMirror Version**: 2.3.1

**Description**: "Feels like" heat index on the currentweather module is reporting wildly inaccurate values in Fahrenheit.

[image]

**Steps to Reproduce**: Use the following currentweather config and hope that it's still hot outside. :)
```
                                location: "McKinney",
                                locationID: "4710178",  //ID from http://www.openweathermap.org/help/city_list.txt
                                appid: "e31cba4d5d9b70891e838dbf9a1b4b64",
                                roundTemp: true,
                                appendLocationNameToHeader: false
```

**Expected Results**: A (more or less) accurate "feels like" temperature.

**Actual Results**: Fatal heat levels.

**Configuration**: See 'steps to reproduce'.

nhubbard, 2018-04-25:
@parnic Good PR! I will approve momentarily. This causes some unbelievably funny issues without your fix, especially in Kelvin (`default`).

nhubbard, 2018-04-25:
Specifically, temperatures of nearly 1500 degrees outside.

parnic, 2018-04-25:
I didn't modify how it was treating Kelvin, so an additional change may be necessary.

MichMich, 2018-05-01:
Merged.

PR #1264 (2018-05-01): Fixed heat index
Celsius and Fahrenheit were flipped. The index was computed in Fahrenheit but used as if it were Celsius.

Fixes #1263
closed 2018-05-01 by MichMich