MagicMirrorOrg/MagicMirror #1266: Getting a init.js file not found

steveboatface, 2018-04-19:
Myapologies on this, i'm not sure where to go with this, as i know it seems like an Electron error based on the file not found below path, but I wasn't sure if this is maybe from your package building the electron pre-requisites or not. So i thought I would give it a shot. This is actually in the node-helper when i make this call:

req.jsonBody = JSON.parse(data);

It traces into the event emitter and then suddenly it gives me this error. Any thoughts?

Unable to open 'init.js': File not found (file:///c:/Users/stephen.botha/AppData/Roaming/npm/node_modules/electron/dist/resources/electron.asar/browser/init.js).

nhubbard, 2018-04-25:
@steveboatface I would try backing up your configuration and reinstall Raspbian. This is an error with Electron.

MichMich, 2018-05-01:
As nhubbart suggests ...

closed 2018-05-01 by MichMich