MagicMirrorOrg/MagicMirror #1267: My Google iCal failed to sync

werdnllac, 2018-04-19:
Please only submit reproducible issues. 
In my config.js, the script is as follow:

```
		{
			module: "calendar",
			header: "Upcoming Appointments",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check-o ",
						url: "webcal://calendar.google.com/calendar/ical/callndrew%40gmail.com/private-ee80c0192dec18c145a87e121205194c/basic.ics"
					}
				]
			}
		},
```
It use to work fine until lately, I decide to upgrade Raspbian Jessie Lite to Raspbian Full version. It somehow couldn't retrieve the Google iCal as shown in the screen shot. 

`I tried both url: "webcal://calendar.google.com/calendar/ical/MyGmail/private-ee80c0192dec18c145a87e121205194c/basic.ics" `

**OR**

`url: https://calendar.google.com/calendar/ical/MyGmail/private-ee80c0192dec18c145a87e121205194c/basic.ics"`

It use to work till I change the Raspbian version to stretch full with Desktop. Anyone knows what is the problem?

[image]

drftg, 2018-04-20:
You might want to edit this post as it contains private information. 
But that made it possible to see that your calendar generates the following error

Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100 (CET) are not supported
 
You should remove those appointments if you want it to work with MM.

werdnllac, 2018-04-24:
@drftg may I know where did you get the "Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100(CET)"

I had changed the Raspi Preference local and date setting and also the raspi-config. They all looks fine.

I have also checked the google calendar, they are also also correct.

drftg, 2018-04-24:
If you ssh into your pi 
```
cd ~/MagicMirror
npm start
```
you will see all logging of MagicMirror

werdnllac, 2018-04-24:
I have execute the above command and have the following screenshot.

[image]
[image]

werdnllac, 2018-04-24:
The strange thing is, when I add a brand new calendar in Google, and uses the public sharing url, it works fine. Don't understand why and have gone through all the setting but still no clue how to resolve.

MichMich, 2018-05-01:
The simplest suggestion is to do a reinstall of Raspbian full edition. It might have something to do with a ssl connection which is out of scope of the MM project.

anooproy, 2018-05-03:
@werdnllac  Hello sir can u get me calendar module of your version.

MichMich, 2018-06-27:
Can I close this issue?

bbrendon, 2018-09-27:
This happens for me too. Can't the module just ignore invalid calendar items?

closed 2019-07-09 by stale[bot]

axe1OO4, 2021-02-10:
Hi, im having issues with the schedule in my google calendar connected to the mmm-calendarext. Can someone help me how to show the schedule in the monthly calendar chart? Thanks

MichMich, 2021-02-10:
Please use the forum for support requests: http://forum.magicmirror.builders