MagicMirrorOrg/MagicMirror #1272: Need Correct and Running code

liquid1, 2018-04-24:
anyone who have already created and is in running form, kindly send the code.
because am facing alot of issues in these modules( weather forecast, weather, facial recognition) & if changing other default modules.
i have placed issues here but no response.
i have been working on this project for 2 months, kindly help me out

justjim1220, 2018-04-25:
can you post your config.js and/or your log files.?
Or, send them to me through private chat?

MichMich, 2018-05-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-05-01 by MichMich

liquid1, 2018-05-21:
hi jim, sorry for late emailing actually alot late.. i was away..
dude can you navigate me the config file??

liquid1, 2018-05-21:
aahh thanks man, got it