MagicMirrorOrg/MagicMirror #1280: Can pm2 auto start MagicMirror2 on startup in node serveronly mode?

hsukup1, 2018-05-05:
**Platform**: RPi3 B, Chromium Browser, Raspbian Jessie Stretch Full Version

**Node Version**:  v9.11.1

**MagicMirror Version**: MagicMirror v2

**Description**: I want to run MM2 from the Chromium browser on my RPi3-B, since I can then be logged into my google account and display my Google Calendar (via MMM-iFrameReload...it works) without having to add in OAuth2.0 support to the module myself. I have used pm2 to autostart the MagicMirror app and that is nice....but my Google account info will not show on the MM this way. 

How can I have the node serveronly command automatically start then stay running? I am connecting (currently) via SSH (PuTTy) but have also been directly at the RPi3 running MM2 and same result. As soon as I close the terminal window, the server hangs up and can no longer refresh/display from Chromium browser.

**Configuration**: Configuration is fine

hsukup1, 2018-05-05:
SOLVED (And sorry for the inconvenience).

Here is the link to the original issue and the solution...
https://stackoverflow.com/questions/12701259/how-to-make-a-node-js-application-run-permanently#comment73761740_12701259

As you will see from the post in the above link, the answer was to run it like this:

node ~/MagicMirror/serveronly > stdout.txt 2> stderr.txt &

Thanks to [DogNibbler](https://stackoverflow.com/users/2469227/dognibbler) on StackDriver.

closed 2018-05-05 by MichMich

hsukup1, 2018-05-05:
ISSUE 2 SOLVED 

From terminal, typed:
`netstat` `-np
to get the PID of my node process
then typed:
`kill` 12345 //where 12345 is the PID of your node `instance
also can find PID of node instance with this command:
`ps` `-A
but that does not return as much information in case you need to verify the IP:port of the node instance before killing it.

NOW...to run MagicMirror2 as a server only using pm2....

in the terminal, enter the following:
`pm2-runtime` ~/MagicMirror/serveronly > stdout.txt 2> stderr.txt &
then enter:
`pm2` list
[image]

Isn't that gorgeous!!!

Now to autorestart the serveronly after a crash or pi reboot.... enter:
`pm2` save

caphill7, 2019-05-31:
Before finding this post (I was actually looking for something else), I solved this problem by not using 'pm2' at all, and instead created 'systemd' services. This way there is full control over start, stop, restart. Note that this is for using MM as 'serveronly' and not the electron app. Combined with other tools, such as 'supervisor' for example, you can get more granularity and even have the client browser to refresh the page after the service restarts. This solution was based on work created by another MM user, which can be found here: https://github.com/alexiri/MagicMirror/compare/master...alexiri:reload