MagicMirrorOrg/MagicMirror #1281: Unable to install dependencies!

xuyichenmo, 2018-05-06:
i run the setup automanticlly, but when it comes to this part, i had an error
      > node ./download-chromedriver.js
it warns that
---------------------------------------------------------
/home/pi/MagicMirror/node_modules/spectron/node_modules/electron-chromedriver/download-chromedriver.js:16
  if (error != null) throw error
                     ^

Error: read ECONNRESET
    at TLSWrap.onread (net.js:602:25)
npm WARN grunt-stylelint@0.10.0 requires a peer of stylelint@^9.0.0 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@1.7.1 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the electron-chromedriver@1.7.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-05-06T08_09_20_207Z-debug.log
-------------------------------------------------------------
it seems that i lost my grunt, but when i try to fix my grunt, it warns that
-------------------------------------------------------------
pi@raspberrypi:~ $ npm install --save-dev grunt
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

+ grunt@1.0.2
updated 1 package in 6.218s
-------------------------------------------------------------
until now i still stuck in this question, help!

MichMich, 2018-05-11:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-05-11 by MichMich

mustafaevrenpolatsamm, 2018-06-19:
I have the same error message

Please help!