MagicMirrorOrg/MagicMirror #1285: UnhandledPromiseRejectionWarning (after MMM-Remote-Control module setup)

hsukup1, 2018-05-10:
**Platform**: Chromium (latest), RPi3B

**Node Version**: 9.11.1

**MagicMirror Version**: v2

**Description**:
Getting error from MMM-Remote-Control module, as well as MMM-ModuleScheduler config'ed to turn on/off monitor via MMM-Remote-Control payload not working. MMM-Remote-Control URL does work to manually turn it on/off on the URL.

**Steps to Reproduce**: 
1) Install MMM-Remote-Control:
`cd ~/MagicMirror/modules`
`git clone https://github.com/Jopyth/MMM-Remote-Control.git`
`cd MMM-Remote-Control`
`npm install`

2) Add module to ~/MagicMirror/config/config.js and update MMM-ModuleScheduler config to turn monitor on/off:
```
                {
                        module: "MMM-ModuleScheduler",
                        config: {
                               notification_schedule: [
                                        {notification: 'REMOTE_ACTION', schedule: '00 11 * * *', payload: {action: 'MONITOROFF'}},
                                        {notification: 'REMOTE_ACTION', schedule: '45 6 * * *', payload: {action: 'MONITORON'}},
                                ],
                        }
                },
                {
                        module: 'MMM-Remote-Control',
                        position: 'bottom_right'
                },
```
3) Run as server:
`cd ~/MagicMirror && node serveronly`

**Expected Results**: 
smooth sailing

**Actual Results**: 
1) [image]
2) Viewing MagicMirror in Chromium browser on RPi3...remote actions not turning HDMI on/off as config'ed in MMM-ModuleSchedule, but does turn HDMI on/off from remote URL given in MMM-Remote-Controle module.

**Configuration**: see above

MichMich, 2018-05-11:
Please create an issue in the repository of this third party module. Thanks.

closed 2018-05-11 by MichMich

d-Rickyy-b, 2018-10-25:
I got a pretty similar error log:

```
(node:8419) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property 'hash' of null
(node:8419) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:8419) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): TypeError: Cannot read property 'hash' of null
(node:8419) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 3): TypeError: Cannot read property 'hash' of null

...
(node:8419) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 486): TypeError: Cannot read property 'hash' of null
(node:8419) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 487): TypeError: Cannot read property 'hash' of null
```

I cannot see that the issue is related to a specific module from the log alone. (**PS**: _Could you please add timestamps to the logging? This is really a mess to debug - I can't correlate the events with other information I get because of missing timestamps_).

@MichMich could you please help me to identify the source/module where the exception is thrown? Do you need an own issue for this? My guess would be that this happens when the RasPi (connected via WiFi) loses the connection.

Greetings, Rico

Jopyth, 2018-11-19:
For further discussion: https://github.com/Jopyth/MMM-Remote-Control/issues/87