MagicMirrorOrg/MagicMirror #1288: Using MagicMirror when Github is blocked

jinsong04, 2018-05-13:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 3

**Node Version**: 8.2.1

**MagicMirror Version**: v2

**Description**: 
The place where I live in has a very strict firewall. OpenVPN doesn't work, packet obfuscation does not work either. Using Astrill, I was able to share the astrill connection from my computer to the raspberry pi to install magicmirror and get it started. Then, I imported all the modules that I want from github to bitbucket and pulled from the pi. Just today, I realized that all the modules won't update without github and it spits this out.

`fatal: unable to access 'https://github.com/MichMich/MagicMirror.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/ronny3050/email-mirror.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/raywo/MMM-NowPlayingOnSpotify/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/frdteknikelektro/MMM-SimpleLogo/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/frdteknikelektro/MMM-SimpleLogo/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/MichMich/MagicMirror.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/raywo/MMM-NowPlayingOnSpotify/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

fatal: unable to access 'https://github.com/ronny3050/email-mirror.git/': server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
`
Is there any way to skip GitHub, or use bitbucket for this? Any help is appreciated. Thank you.

fewieden, 2018-05-13:
for me it sounds like the updatenotification module, you don't need that to run the mirror, you can disable it in the config

jinsong04, 2018-05-14:
Oh that's what it was. Thank you very much!

closed 2018-05-15 by MichMich