MagicMirrorOrg/MagicMirror #1291: Documentation: Unit-Tests for 3rd Party Modules

ArneCL, 2018-05-16:
Unfortunately, there is no documentation on how to write unit-tests for 3rd-party modules. 
There are unit-tests for the default-modules, however, I've not yet been able to port them to a 3rd-Party module :(
How would you access the functions of a 3rd party module in a unit test? 

1.  ModuleFile, i.e. `Module.register("myModule", { func1:  ... })`

(The following did not work for me, taken from default-module tests)
```Module = {}
Module.definitions = {};
Module.register = function (name, moduleDefinition) {
	Module.definitions[name] = moduleDefinition;
};
require("myModule");
```

2. NodeHelper, i.e. `module.exports = NodeHelper.create({ func1: ... })`
Here an approach could be to stub `NodeHelper` _before_ the module under test requires it - which seems a bit fragile to me.

A forum-search did not reveal any information and I think this could potentially be of interest to other module-developers. Your help/suggestions are welcome

MichMich, 2018-05-17:
Which "the third-party module template" do you mean?

ArneCL, 2018-05-17:
The [one](https://github.com/roramirez/MagicMirror-Module-Template) mentioned [here](https://github.com/MichMich/MagicMirror/wiki/3rd-party-modules) (Does not seem to get too much attention recently). I am curious if you would rather suggest to mock/stub Classes like NodeHelper (wich would increase complexity due to tools like _proxyquire_, but make the tests run for themselves) **or** require them from the MMM-Environment (wich would make test less unit-focused). The last (as done by [ModuleScheduler](https://github.com/ianperrin/MMM-ModuleScheduler) and in the default modules) of course also makes sense, as a vital change in e.g. NodeHelper  would cause the module-tests to fail (which is a good thing, I guess)...

I would be willing to create a PR for the [MagicMirror-Module-Template](https://github.com/roramirez/MagicMirror-Module-Template) including your thoughts on this. However, now realising that is not your repo and did not seem to get a lot of attention recently, don't really now if it will get merged ....

MichMich, 2018-05-17:
That template is not part of the official repo. I agree it would make sense to add the test to that template, but I'd advice you to open an issue in that repo. (It's not maintained by me).

The repo is maintained by @roramirez, who is also one of the admins of the forum (so an active community manager) and the person who added all the tests to the core anyway.

MichMich, 2018-05-21:
This this regards an other repository, I close this issue. Feel free to reopen if necessary.

closed 2018-05-21 by MichMich

roramirez, 2018-05-23:
Hi guys! @MichMich and @ArneCL I'll take a look you if you need want push and add unit-test the 3rd-party modules. I think is a good point give this feature to other modules to complement inside of MagicMirror.

@ArneCL Let me know if need something.