MagicMirrorOrg/MagicMirror #1292: "invalid date" within weather forecast

t4rif4ri, 2018-05-20:
**Platform**: rpi 3, newest pull

**Description**: Weather forecast is showing just one forecast like this "invalid date, then max and min. temperature.  
No forecast for 7 days like it used to be before.

**Configuration**: forecast configuration contains just the location, locationid and appid.

MichMich, 2018-05-21:
Thanks for opening an issue!

To allow us to help you it's extremely important that we have all the details about your setup. During the creation of this issue, important details were asked to fill in. Because you did not provide (any) of these details we are unable to help you. 

If you are running into errors, please provide as many detailed error logs as possible in addition to the requested details.

If you are unaware how to retrieve all the requested information, feel free to ask for help on http://forum.magicmirror.builders.

This issue will be closed. Feel free to open a new issue with all the requested information.

closed 2018-05-21 by MichMich

Canterrain, 2018-06-24:
Hello!
I'm experiencing exactly the issue described above.

I posted some pm2 logs and additional details here:
https://forum.magicmirror.builders/topic/7459/weather-forecast-only-displays-invalid-date

but nothing about the logs even seems to be suggesting an issue.  It's weird.  Anything else I can provide?