MagicMirrorOrg/MagicMirror #1294: Calendar not respecting timeFormat: "absolute" for full-day events

michaelfisher, 2018-05-24:
Platform: Raspberry Pi 3

Node Version: v9.11.1

MagicMirror Version: v2.3.1

Description: The calendar module displays full-day events as relative regardless of timeFormat: "absolute" , getRelative: 0, urgency:0, and fullDayEventDateFormat: "ddd".

Screenshot:
[image]

Steps to Reproduce: Create a full-day event, and have the above listed config options set.

Expected Results: I expect that the full-day event will be displayed with the specified fullDayEventDateFormat.

Actual Results: Full-day events are displayed as relative (e.g. "In 2 Days").

Configuration: 

module: "calendar",
	header: "3 - Day Overview <span class='fa fa-fw fa-calendar'></span>",
        position: "top_right",
        config: {
        	maxTitleLength: 20,
                fetchInterval: 60000,
                animationSpeed: 0,
                fade: false,
                dateFormat: "ddd[,] h:mm A",
                fullDayEventDateFormat: "ddd",
                timeFormat: "absolute",
                getRelative: 0,
                urgency: 0,
                calendars: [
                	{
                        	//Personal
                                symbol: "user ",
                                color: "#5DADE2",
                                url: "https://calendar.google.com/calendar/ical/blah-blah-blah
                                maximumNumberOfDays: 3
                        }

Additional Notes: The events are created in Google calendar, and here is the section containing the event from the ICS file

BEGIN:VEVENT
DTSTART;VALUE=DATE:20180524
DTEND;VALUE=DATE:20180525
DTSTAMP:20180523T235527Z
UID:XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX (wasn't sure if this was sensitive info)
CREATED:20180523T235500Z
DESCRIPTION:
LAST-MODIFIED:20180523T235500Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Test Full-Day Event
TRANSP:OPAQUE
BEGIN:VALARM
ACTION:NONE
TRIGGER;VALUE=DATE-TIME:19760401T005545Z
END:VALARM
END:VEVENT

lavolp3, 2018-11-21:
I can reproduce this only when my mirror is set to 'en' and for the day after tomorrow
There is a section handling the translation of DAYAFTERTOMORROW.
Oddly, in English it seems to substitue dayaftertomorrow with the relative date. That doesn't make sense.

ABOATDev, 2018-11-29:
I have the same problem 
 
When I record an event for a full day for tomorrow (via Google calendar)

Calendar Magic Mirror displays it in 2 days

lavolp3, 2018-11-30:
> I have the same problem
> 
> When I record an event for a full day for tomorrow (via Google calendar)
> 
> Calendar Magic Mirror displays it in 2 days

Well, that's not the same problem.
The original issue was not showing the event on a wrong date, but with relative format albeit absulote was set.
For your problem you would need to go a bit more into detail.
At which time did you set the calendar, which time of the day have you seen the issue.
What exactly does the calendar show?

closed 2019-07-09 by stale[bot]