MagicMirrorOrg/MagicMirror #1295: Problem with the new privacy rules

el97, 2018-05-25:
Hi! When i tried to login to the forum i got stuck on the personal info site. there is a box that says "You have not provided consent for data collection and processing. At any time this website's administration may elect to delete your account in order to become compliant with the General Data Protection Regulation." and also a button that says "Your rights & consent". But when i click on the button the page just reloads and the same thing happens again. This might not be the correct place to put this on but i have nowhere else to go...

MichMich, 2018-05-25:
@paviro. Can you take a look?

paviro, 2018-05-25:
On it. Does it work for you @MichMich?

MichMich, 2018-05-25:
Yeah. No issues.

fewieden, 2018-05-26:
One user reported on the forum

`Obviously I figured out a way (or I wouldn’t be posting now 😉 ) but I had to delete all the cookies for the forum and log back in before I could actually use the forum.`

and for me `a simple f5 solved the issue`

https://forum.magicmirror.builders/topic/7199/consent-loop

el97, 2018-05-28:
It seems like i fixed my problem. The problem only occurs when i tried to login with github so i guess this ticket should be open until further notice.

closed 2018-06-06 by MichMich