MagicMirrorOrg/MagicMirror #1298: Cannot render footer in the forum

fewieden, 2018-05-27:
By accessing any page in the forum, it only displays:

```
Render failed for template /home/forge/forum.magicmirror.builders/build/public/templates/footer.js:
 TypeError: Render failed for template /home/forge/forum.magicmirror.builders/build/public/templates/footer.js:
 templateFunction is not a function
    at runtime (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/runtime.js:64:16)
    at render (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/express.js:11:20)
    at Object.__express (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/express.js:33:5)
    at /home/forge/forum.magicmirror.builders/src/webserver.js:157:15
    at async.waterfall.Benchpress.precompile.source (/home/forge/forum.magicmirror.builders/src/middleware/index.js:226:12)
    at nextTask (/home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:5310:14)
    at next (/home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:5317:9)
    at /home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:958:16
    at /home/forge/forum.magicmirror.builders/src/file.js:146:3
    at /home/forge/forum.magicmirror.builders/node_modules/graceful-fs/polyfills.js:287:18
    at FSReqWrap.oncomplete (fs.js:123:15)
```

el97, 2018-05-27:
Same here. What browser do you use?

fewieden, 2018-05-27:
It's not browser related, that's a server issue.

MichMich, 2018-05-27:
@paviro

weaklion, 2018-05-28:
i want get solution.

fewieden, 2018-05-28:
works again

Wolfieeewolf, 2018-05-28:
Broken again

Render failed for template /home/forge/forum.magicmirror.builders/build/public/templates/footer.js:
 TypeError: Render failed for template /home/forge/forum.magicmirror.builders/build/public/templates/footer.js:
 templateFunction is not a function
    at runtime (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/runtime.js:64:16)
    at render (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/express.js:11:20)
    at Object.__express (/home/forge/forum.magicmirror.builders/node_modules/benchpressjs/build/lib/express.js:33:5)
    at /home/forge/forum.magicmirror.builders/src/webserver.js:157:15
    at async.waterfall.Benchpress.precompile.source (/home/forge/forum.magicmirror.builders/src/middleware/index.js:226:12)
    at nextTask (/home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:5310:14)
    at next (/home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:5317:9)
    at /home/forge/forum.magicmirror.builders/node_modules/async/dist/async.js:958:16
    at /home/forge/forum.magicmirror.builders/src/file.js:146:3
    at /home/forge/forum.magicmirror.builders/node_modules/graceful-fs/polyfills.js:287:18
    at FSReqWrap.oncomplete (fs.js:123:15)

fewieden, 2018-05-28:
just checked it, seems to work again :D

roramirez, 2018-05-28:
I did a rebuild of assets files in the  server. One hour before, and apparently is working now. Please, check is work for you. Thanks for report.

el97, 2018-05-28:
It works for me now.

MichMich, 2018-05-29:
@Wolfieeewolf Does it work for you now?

Wolfieeewolf, 2018-05-29:
Working again :) Sorry working night shifts and couldn't reply until now.

MichMich, 2018-05-29:
Thanks! Closing topic.

closed 2018-05-29 by MichMich