MagicMirrorOrg/MagicMirror #1307: Brithday Calendars with Birthdays prior to 1970 are not shown.

TheRealBrother, 2018-06-06:
**Platform**: Raspberry Pi 3

**Node Version**: v9.10.1

**MagicMirror Version**: newest

**Description**: Using a Birthday Calendar with Birthdays beeing prior to 1970 the following error occurs when importing:

 Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100 (CET) are not supported

**Steps to Reproduce**: Create a Calendar in nextcloud, with contact brithdays that are not millennials ;)
Export that calendar as ical

**Expected Results**: Birthdays are shown in calendar view

**Actual Results**: No Birthdays are shown

**Configuration**: 
iCal config part:
{
symbol: "calendar",
url: "https://user:password@my.domain/cloud/remote.php/dav/calendars/[username]/contact_birthdays/?export"
}

MichMich, 2018-06-06:
See introduction text:
https://github.com/MichMich/MagicMirror/tree/develop/modules/default/calendar#module-calendar

closed 2018-06-06 by MichMich

DraakUSA, 2018-06-07:
The reason for this "breakdown" is that time [in iCal and other libraries] is usually computed using the standard Unix Epoch time, which is the number of seconds since Jan 01 1970 UTC.   This is not the fault of the MagicMirror code.

DraakUSA, 2018-06-07:
I dug a bit more into where the error is occurring.  rrule.js is the module throwing "Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100 (CET) are not supported".  

If you change line 70 of MagicMirror/package.json to:

    "rrule-alt": "^2.2.8",

Then it will allow dates down to Jan 01 1900 UTC