MagicMirrorOrg/MagicMirror #1308: module.suspend not called if module does not have dom content

sdetweil, 2018-06-07:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Ubuntu Mate on ODROID xu4

**Node Version**: v6.14.2

**MagicMirror Version**: 2.3.1

**Description**: my modules, which have no dom content (and no position specified in config) do not receive the suspend notification. 

**Steps to Reproduce**: create the default module as defined in the doc, but only supply the  module name in config.js 
at this point issue a module.hide() against this module, but its suspend method will not be called. 

**Expected Results**: module suspend() function should be called

**Actual Results**: suspend not called (resume is)

**Configuration**: ```
{
                                module: "ImageScheduler",
                },
                {
                                module: "ImageService",
                },
```

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

this is due to the test for dom content in js/main.js/hideModule()
```
	var moduleWrapper = document.getElementById(module.identifier);
        if (moduleWrapper !== null) {
```
there should be an else, to invoke the (post fade timeout)callback setup in js/module.js/hide()
```
	}
	else
	     if (typeof callback === "function") { callback(); }
```

MichMich, 2018-06-07:
Thanks. Feel free to send a PR.

sdetweil, 2019-02-20:
pr merged july 2018, 
https://github.com/MichMich/MagicMirror/pull/1309

closed 2019-02-20 by sdetweil