MagicMirrorOrg/MagicMirror #1313: How to uninstall auto start magic mirror

SaraPostma, 2018-06-08:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

justjim1220, 2018-06-08:
have you tried the Alt button?
it should bring up a bar that will have this at the top...
[image]

then click on File and click Exit.

JiJoker1991, 2018-06-18:
1. Open Terminal
2. pi@raspberrypi:~ $ pm2 stop MagicMirror

or

pi@raspberrypi:~ $ pm2 stop mm

sharmstr, 2018-06-18:
As JiJoker1991 mentioned, you can use the stop command to stop mm from restarting until you reboot, but if you really want to "uninstall auto start" you need to use the delete command.

Open terminal
type in "pm2 list" to see if your magic mirror pm2 process is called MagicMirror or mm.
type in "pm2 delete xxx"  where xxx is either MagicMirror or mm.

wast, 2018-06-21:
"pm2 kill" will remove all processes from its startup list.

MichMich, 2018-06-27:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-06-27 by MichMich

lee-lvd, 2019-01-16:
all i ever see is the Msgic mirror 
I can no longer get to my gui screen log in
i like this but do wish I could get back to that screen