MagicMirrorOrg/MagicMirror #1325: Clock module : bold is only bolding the minute

prysme01, 2018-06-20:
Hello, 

After setting the parameter to bold at true, I only see the minute "bolded" not the hours.

thanks.

prysme01, 2018-06-20:
As I can see in the code 
`		if (this.config.clockBold === true) {
			timeString = now.format(hourSymbol + "[<span class=\"bold\">]mm[</span>]");
		} else {
			timeString = now.format(hourSymbol + ":mm");
}`

Is this the expected behavior to only bold the minute ?

nhubbard, 2018-06-20:
Yes. It is intended to only bold the minute and remove the colon to produce a more modern look. See this screenshot for the intended result:
[image]

(Source: am author of bold option in clock module.

prysme01, 2018-06-21:
Well on my installation with no special customization in the CSS, the expected result is not the same
[https://framapic.org/eAoARpyqQbu5/SOS8J17z4RnU.jpg](url)

justjim1220, 2018-06-23:
this is where I changed it to get the full time to be bold... in the **_clock.js_** file
`// Style Wrappers
		dateWrapper.className = "date normal xlarge bolder";
		timeWrapper.className = "time bright xlarge bolder";
		secondsWrapper.className = "dimmed";
		weekWrapper.className = "week dimmed medium"
`

justjim1220, 2018-06-23:
[image]

closed 2018-06-27 by MichMich