MagicMirrorOrg/MagicMirror #1327: no helper found for modules

onkararj, 2018-06-21:
> magicmirror@2.1.2 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.1.2
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Launching application.
[image]
[image]

fewieden, 2018-06-21:
not all modules require a node helper and most of the default modules have one.

onkararj, 2018-06-23:
so what i have to do? to turn on all the modules

fewieden, 2018-06-23:
Just saying that this is not your issue, those are just informations that those modules do not have a node_helper. It's not the reason why your screen is black. Open the developer tools in electron and check the console for errors.

justjim1220, 2018-06-23:
you need to update Magic Mirror, you're using version 2.1.2, and the current version is 2.3.1

MichMich, 2018-06-27:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-06-27 by MichMich

wajidraza786, 2019-11-25:
try this command in module folder:
npm i

chiliwax, 2020-11-08:
`npm run install` fix the problem for me (install fonts and usefull stuffs to display modules)
MM2 version : `2.14.0-dev`
Run on RPI 1.2 (raspberry 1 b+)