MagicMirrorOrg/MagicMirror #1329: "Can't parse feed item:" flooding log, disable this or make a debug level

flyingchipmunk, 2018-06-24:
**Platform**: Raspberry Pi 3B+, electron 1.7.13

**Node Version**: v9.11.2

**MagicMirror Version**: v2

**Description**: The ~/.pm2/logs/MagicMirror-out-0.log is getting flooded with "Can't parse feed item:" messages from the default/newsfeed module. Regardless of source of the newsfeeds that can't be parsed this message needs to be removed or set to a debug level and a flag to enable it. On a production installation this should not be printing as console.log(). On an unchecked system this will flood the log file and speed the destruction of the sdcard with unneeded writes.

**Steps to Reproduce**: Leave the default NYT RSS feed that fetcher.js can't parse, i.e. http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml and watch the log for the errors.

**Expected Results**: No log messages written unless a fatal operation has occurred.

**Actual Results**: Message like this flood the out.log
```
Can't parse feed item:
{ title: '',
  link: 'https://www.nytimes.com//?partner=rss&emc=rss',
  guid: { ispermalink: 'true', text: 'https://www.nytimes.com//' },
  'atom:link':
   { rel: 'standout',
     href: 'https://www.nytimes.com//?partner=rss&emc=rss' },
  description: '',
  'dc:creator': '',
  pubdate: '' }
Title:
Description:
Pubdate: undefined
```

**Configuration**: 
```
        {
            module: "newsfeed",
            position: "bottom_bar",
            config: {
                feeds: [
                    {
                        title: "New York Times",
                        url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                    }
                ],
                showSourceTitle: true,
                showPublishDate: true
            }
        },
```

**Additional Notes**:

almazini, 2018-06-26:
I met the same issue today on Raspberry 3 today

MichMich, 2018-06-26:
Feel free to send a PR to disable this using a configuration option.

flyingchipmunk, 2018-06-26:
I briefly looked at how you implemented the logger class on the non-module side of the code. Didn't know if there were plans to expand that to modules. If not, and a config option for this module would be acceptable for this I can work up a PR for it.

MichMich, 2018-06-26:
Yeah for now a config option would be the easiest solution. Thanks!

MichMich, 2018-06-27:
Merged

PR #1331 (2018-06-27): Add option to newsfeed for logging errors
- 'logFeedWarnings' added to newsfeed config, defaulted to false
- Only log parse feed errors when logFeedWarnings is true
- Updated README and CHANGELOG
- Fixes #1329
commit e5637711 (2018-06-27): Add option to newsfeed for logging errors

- 'logFeedWarnings' added to newsfeed config, defaulted to false
- Only log parse feed errors when logFeedWarnings is true
- Updated README and CHANGELOG
- Fixes #1329
closed 2018-06-27 by MichMich