MagicMirrorOrg/MagicMirror #1332: "Invalid date" within weather forecast

Canterrain, 2018-06-27:
This seemed to start after performing latest update. Weather forecast had been working for me for a long time (I built the mirror at least two or three years ago).

Logs don't seem to reveal much:
`

mm-0 Starting MagicMirror: v2.4.0-dev
mm-0 Loading config ...
mm-0 Loading module helpers ...
mm-0 No helper found for module: alert.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: MMM-Motion-Detection
mm-0 No helper found for module: clock.
mm-0 No helper found for module: MMM-Nest.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: calendar
mm-0 No helper found for module: currentweather.
mm-0 No helper found for module: weatherforecast.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: newsfeed
mm-0 All module helpers loaded.
mm-0 Starting server on port 8080 ...
mm-0 Server started ...
mm-0 Connecting socket for: MMM-Motion-Detection
mm-0 Starting module helper: MMM-Motion-Detection
mm-0 Connecting socket for: calendar
mm-0 Starting node helper for: calendar
mm-0 Connecting socket for: newsfeed
mm-0 Starting module: newsfeed
mm-0 Sockets connected & modules started ...
mm-0 Launching application.
mm-0 Create new calendar fetcher for url: h
mm-0 Create new news fetcher for url: http://feeds.arstechnica.com/arstechnica/index/ - Interval: 300000
mm-0 Create new news fetcher for url: http://www.npr.org/rss/rss.php?id=1001 - Interval: 300000
mm-0 [MMM-Motion-Detection] Motion Detection started...
mm-0 [MMM-Motion-Detection] --------------------
mm-0 [MMM-Motion-Detection] Webcam starting...
mm-0 [MMM-Motion-Detection] motion detected
mm-0 Starting MagicMirror: v2.4.0-dev
mm-0 Loading config ...
mm-0 Loading module helpers ...
mm-0 No helper found for module: alert.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: MMM-Motion-Detection
mm-0 No helper found for module: clock.
mm-0 No helper found for module: MMM-Nest.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: calendar
mm-0 No helper found for module: currentweather.
mm-0 No helper found for module: weatherforecast.
mm-0 Initializing new module helper ...
mm-0 Module helper loaded: newsfeed
mm-0 All module helpers loaded.
mm-0 Starting server on port 8080 ...
mm-0 Server started ...
mm-0 Connecting socket for: MMM-Motion-Detection
mm-0 Starting module helper: MMM-Motion-Detection
mm-0 Connecting socket for: calendar
mm-0 Starting node helper for: calendar
mm-0 Connecting socket for: newsfeed
mm-0 Starting module: newsfeed
mm-0 Sockets connected & modules started ...
mm-0 Launching application.
mm-0 Create new news fetcher for url: http://feeds.arstechnica.com/arstechnica/index/ - Interval: 300000
mm-0 Create new news fetcher for url: http://www.npr.org/rss/rss.php?id=1001 - Interval: 300000
mm-0 [MMM-Motion-Detection] Motion Detection started...
mm-0 [MMM-Motion-Detection] --------------------
mm-0 [MMM-Motion-Detection] Webcam starting...
mm-0 [MMM-Motion-Detection] motion detected`

and config doesn't show any problems with the config checker:
`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
	timeFormat: 12,
	units: 'imperial',

	modules: [
		{
			module: 'alert',
		},
		{
			module: 'MMM-Motion-Detection,'
		},
		{
			module: 'clock',
			position: 'top_left'
		},
		{
			module: 'MMM-Nest',
			position: 'bottom_right',
			config: {
				token:"xxx"
			}
		},
		{
			module: 'calendar',
			header: 'Calendar',
			position: 'top_left',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'xxx'
					}

				]
			}
		},
		
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: 'xxx',
				locationID: 'xxx',  //ID from http://www.openweathermap.org
				appid: xxx'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			config: {
	            location: 'xxx',
				locationID: 'xxx',  //ID from http://www.openweathermap.org
	            appid: 'xxx'
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_bar',
			config: {
				feeds: [
					{
						title: "ArsTechnica",
						url: "http://feeds.arstechnica.com/arstechnica/index/",
					},
					{
						title: "NPR",
						url: "http://www.npr.org/rss/rss.php?id=1001"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
`

gberg927, 2018-07-02:
I'm also seeing this issue. Thanks!

vincep5, 2018-07-02:
Same issue for me today @berlincount .  I am noticing the API json does not have dt_txt.  
https://api.openweathermap.org/data/2.5/forecast/daily?id=xxxxx&units=imperial&lang=en&APPID=xxxxx
fill in xxxxx with your values

GregoryOrciuch, 2018-07-02:
Just updated with git pull to latest 2.4.0. Same for me.

Hilzerak, 2018-07-03:
Same here :/
Any Solutions?

Canterrain, 2018-07-03:
There's a pull request in to correct the problem.  Hopefully that will all clear soon, and then we can update our mirrors confirm the fix and close this issue.

GregoryOrciuch, 2018-07-03:
@Hilzerak solution is pretty manual for now:
you edit this file: modules/default/weatherforecast/weatherforecast.js
and change around lines 336-337 back to:

```js
var day = moment(forecast.dt, "X").format("ddd");
var hour = moment(forecast.dt, "X").format("H");
```

then restart MM.

Hilzerak, 2018-07-03:
@GregoryOrciuch 
Works like a charm. Much thx!

MichMich, 2018-07-03:
Ive just merged the fix by @jannekalliola into the develop branch which should solve this issue: https://github.com/MichMich/MagicMirror/pull/1341

Can you guys please test this branch? If this solves the issue, I'll release an intermediate 2.4.1 release.

To tryout the develop branch, use the following git command on your terminal (from within the MagicMirror folder):
```
git checkout develop 
git pull
```

Thanks for testing!

Canterrain, 2018-07-03:
I believe I'm already on develop branch.  Soon as I get home I'll pull and confirm.

Canterrain, 2018-07-03:
I can confirm that this update fixed this issue!

Thank you everybody!

closed 2018-07-03 by Canterrain