MagicMirrorOrg/MagicMirror #1335: forecast/daily not working

ihorbond, 2018-06-28:
forecast/daily is no longer available as a free service from open weather map
[image]

Jacobphilli, 2018-08-29:
It should work fine still, it just means your max days of forecast is five, even if you have it set to more in the configuration.

And, you want to ensure your reset time is not great than 60 per minute. The default is one call per 15 minutes I believe.

kellnermichael, 2018-10-04:
changing ForcastEndpoint: "forecast/daily" to "forecast" resolved the error I was receiving with max days set to 5 also using location and locationid