MagicMirrorOrg/MagicMirror #1339: Newest MagicMirror Update

meischdaddy30, 2018-07-01:
Hi all,

So an update for MagicMirror just came out today and I have a question about it.  When I navigate to my MagicMirror folder and use the 

`git pull && npm install`

I get the following errors.

`Updating 60b9a5b..3545f80`

`error: Your local changes to the following files would be overwritten by merge:
	package-lock.json
	package.json
	vendor/package-lock.json`

`Please commit your changes or stash them before you merge.
Aborting`

I am still fairly new to writing and interpreting code but I'm wondering what I should do here.  Should I move those files to another directory and retry the update command?  My worry is that if I do this then my MMM-Assistant will not work (it took me about a month to finally get the module to work so I really would not like that to happen).

Thinking to myself it shouldn't affect the MMM-Assistant module at all because these .json files are in my `MagicMirror` directory not my `MMM-Assistant` directory.  But then I just read the updates for the Magic Mirror from [https://github.com/MichMich/MagicMirror/releases](url) it says it has updated the electron version to v2.0.0.  My MMM-Assistant electron-rebuild is currently at v1.7.3 https://github.com/eouia/MMM-Assistant/blob/master/package.json (this is my version as far as I know)

So my real question is, will removing the .json files from the `MagicMirror` directory and then updating the to the newest MagicMirror version affect, in any way, my MMM-Assistant module/directory due to the difference in electron versions?
  
I'm running Raspberry Pi 3 model b
web browser version - Version 65.0.3325.181 (Official Build) Built on Raspbian , running on Raspbian 9.4 (32-bit)
`$ node -v
v9.11.2`
`$ npm -v
6.1.0`

Any help would be much appreciated!

MichMich, 2018-07-01:
https://github.com/michmich/magicmirror#updating

MichMich, 2018-07-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-07-01 by MichMich

meischdaddy30, 2018-07-01:
Okay, thanks!