MagicMirrorOrg/MagicMirror #1344: Upgrade to Font Awesome Version 5

groszejc, 2018-07-04:
I would like to have font awesome version 5 installed within the following file path for additional icons: 

MagicMirror/vendor/node_modules/

This would also need to have the MagicMirror/vendor/vendor.js file updated to reflect the upgrade as well:

var vendor = {
	"moment.js" : "node_modules/moment/min/moment-with-locales.js",
	"moment-timezone.js" : "node_modules/moment-timezone/builds/moment-timezone-with-data.js",
	"weather-icons.css": "node_modules/weathericons/css/weather-icons.css",
	"weather-icons-wind.css": "node_modules/weathericons/css/weather-icons-wind.css",
	**"font-awesome.css": "node_modules/font-awesome_5/css/all.css",**
	"nunjucks.js": "node_modules/nunjucks/browser/nunjucks.min.js"
};

MichMich, 2018-07-04:
I'm not sure how this would affect existing modules. Any idea?

groszejc, 2018-07-04:
I have needed to use the upgraded font awesome symbol specification in my calendar module shown below:

`{
						symbol: "fas fa-calendar-alt",
						url: "webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics",
                        color: "#ffffff"
					},`

The default modules I use have not been affected: Clock, Current Weather, Weather Forecast and News Feed. Unsure of the effects of the Compliments, Hello World or Alert modules.

MichMich, 2018-07-04:
It seems that fa5 isn't backwards compatible to fa4. This means that if we upgrade it, some icons stop working for mirrors that are currently configured. What might be an option, is to make the version of font awesome (in the calendar) configurable, with fa4 as the default.

groszejc, 2018-07-05:
Doing some testing, it looks like we could have both FA4 and FA5 icons available by adding another line to the MagicMirror/vendor/vendor.js file:

var vendor = {
	"moment.js" : "node_modules/moment/min/moment-with-locales.js",
	"moment-timezone.js" : "node_modules/moment-timezone/builds/moment-timezone-with-data.js",
	"weather-icons.css": "node_modules/weathericons/css/weather-icons.css",
	"weather-icons-wind.css": "node_modules/weathericons/css/weather-icons-wind.css",
    **"font-awesome.css": "node_modules/font-awesome_4/css/font-awesome.min.css",
	"font-awesome5.css": "node_modules/font-awesome_5/css/all.css",**
	"nunjucks.js": "node_modules/nunjucks/browser/nunjucks.min.js"
};

Then adding this into the MagicMirror/modules/default/calendar/calendar.js file:

	getStyles: function () {
		return ["calendar.css", "font-awesome.css", "font-awesome5.css"];
	},

This would give the user the ability to select symbols from either FA4 or FA5 within the config.js file, just formatting them correctly as shown below:

```
{
    symbol: "calendar", //FA4 format
    symbol: "fas fa-baseball-ball", //FA5 format
},
```

MichMich, 2018-07-05:
Sounds nice. Can you send a PR to the dev branch? Dont forget the updated changelog. :)

MichMich, 2018-08-16:
Closing dure to inactivity. Feel free to reopen or resubmit a PR.

closed 2018-08-16 by MichMich