MagicMirrorOrg/MagicMirror #1355: Calendar event time is not correct.

NickBeugel, 2018-07-13:
Hi,

I'm having problems with the time of the event its showing in the calendar module.
For example the event is at 19.00, then the modules shows it as 17.00
I thought might be a timezone thing or something  (Amsterdam +2), but the clock module is set to my time zone and displays the right time. 
I'm using Google Calendar for this, but I couldnt find a anything their either.

So if anyone has a solution please let me know!
Thanks

MichMich, 2018-08-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-08-01 by MichMich

AceTheFace, 2019-10-27:
Last night Germany switched back to standard time and now all my events are shown with a wrong begin time (-1h). Timezone of Raspberry and Google Calendar are correct (CET and GMT+1). Before the switch everything was working as expected. The clock module shows the correct date. Any hints how I can fix this offset?

simphide, 2019-10-30:
> Last night Germany switched back to standard time and now all my events are shown with a wrong begin time (-1h). Timezone of Raspberry and Google Calendar are correct (CET and GMT+1). Before the [...]

I have the same problem...

AceTheFace, 2019-11-03:
No, issue still persists. New added events are shown correctly.