MagicMirrorOrg/MagicMirror #1359: Calender

antonstarlid, 2018-07-16:
Option to set when you will update your calender, the "MMM-ModuleScheduler" can do this or if its any other way to do it but i like the way when you can set it like crontab and its updated by the time of your computer and not like "fetchInterval" that i think is counted from the time the MagicMirror is started..

I have 4 different calenders, one is "birthdays" of people i know it only has to be updated at "00:00" every day and so does "weekend" days and "name days" so have a crontab option for those would be awesome lile "0 0 * * *" would do it.

My last calender is more up to date and should be updated alot more often but its only one out of four.

\ Lukasz

antonstarlid, 2018-08-01:
There is a way to look at it if you are ignorant and do not care about others.

I just think it's unnecessary to make these requests so often when you absolutely do not have to do them.

MichMich, 2018-08-01:
Feel free to send a PR to add this functionality using an optional configuration option.

antonstarlid, 2018-08-01:
That´s probably because many companies have an API to be able to control how you use their services and make sure you do not abuse them so even if you can technically make a request in the minute, it's totally unnecessary if you do not need it should be considered as abuse of the service.

jannekalliola, 2018-08-05:
I'm not 100% sure whether I got this right, but there is a fetchInterval configuration value in calendar.js and that is passed to calendarfetcher.js that is doing the actual requesting of the calendars.

If you add the following line to the config.js in calendar configuration:

fetchInterval: 1 * 60 * 60 * 1000

the system will fetch calendars once an hour. This does not help with the quest of having different intervals for each calendars, but it will reduce the number of requests for all of them.

MichMich, 2018-08-16:
Closing this because I don't see any value in this.

closed 2018-08-16 by MichMich