MagicMirrorOrg/MagicMirror #1361: Magic mirror forum

mikecy22, 2018-07-21:
I dont know if it the correct place to post this but i cannot post anything in the forum.
 I have a problem with my confirmation email. I haven't received it. I resend it a couple of times i even changed my email to another one but still havent received anything. My username is mikecy

tuicemen, 2018-07-28:
At least you got a confirmation email.
I've not received one yet.

paviro, 2018-08-01:
mikecy Has already confirmed his email address.

paviro, 2018-08-01:
tuicemen does not exist as a user.

tuicemen, 2018-08-01:
@paviro  I noticed I was approved Thanks.
However I never did get a email, maybe the forum email activation is not working.

paviro, 2018-08-01:
It is. I tested it when I send yours. I received the mail without a problem. Check your spam folders.

mikecy22, 2018-08-01:
Yes i received it 4-5 days later but i was away. Now i am a member!!!

tuicemen, 2018-08-01:
There must be a delay, if mikecy22 email was 4-5 days late I won't see anything for a few more days.
Funny I get the update to this  issue fast though.
Different server?
I don't mind a delay in receiving emails however a post on the forum that registration email activation notifications could take a few days would ease some newbies worries.

mikecy22, 2018-08-01:
I have just checked it. I opened the issue here in github 21 july. The registration email came at 24 of july. So there was some delay.

closed 2018-08-16 by MichMich