MagicMirrorOrg/MagicMirror #1377: Install script error with pm2 autostart

pyrosmiley, 2018-08-10:
Was gonna only make a forum post, but I've run into this with fresh Stretch installs on two separate pis, a 3 and a new 3b. Updated everything then immediately ran MM install script on each, has happened the same each time.

**Platform**: Rpi 3 & 3b, brank spankin' new & updated Raspbian install. Node v9.11.2

**Node Version**: 9.11.2

**MagicMirror Version**: 2.4.1

**Description**:On fresh raspbian installs on each machine, I ran the MM install script, which I have not had issues with before. It actually ran just fine as far as I could tell the first time, but after a reboot of the machine, I noticed the following: MM was not autostarting, I could *start* it from a terminal, but electron never actually opened, and anytime I tried to start it with pm2, the pm2 daemon was not running, and MM was not saved as something to watch.

 It turns out that the `sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"' line was in fact the culprit. when I ran that manually, it gave back an `Unknown platform /init name` error, with the explanation that `PM2 detected systemd but you precised linux`. So I manually ran `pm2 startup` and the command it gave me to input instead was `sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi`. Everything worked fine after that.

**Steps to Reproduce**: Fresh raspbian install, `sudo apt-get update && sudo apt-get upgrade -y`, then execute MM install script from `curl`.

**Expected Results**: I expected it to work like it's supposed to

**Actual Results**: pm2 startup config error

**Configuration**: N/A

**Additional Notes**: I am not sure if an updated version of pm2 or Raspbian or whatever has altered the command for the setup that's used in the script, but it didn't work until I replaced `linux` with `systemd`. I originally had assumed that my pi3 was the culprit (it was one of the ones that ran into the crazy CPU errors with electron, so I know it's already a special boy) but then the same thing happened with a 3b that was fresh out of the box.

Here is the full console output wherein:
1. I attempt to run startup command, and it fails.
2. Startup command successfully set up
3. MM pm2 config successfully loaded

```
pi@mirror:~ $ sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"
[PM2] Init System found: systemd
-----------------------------------------------------------
 PM2 detected systemd but you precised linux
 Please verify that your choice is indeed your init system
 If you arent sure, just run : pm2 startup
-----------------------------------------------------------
/usr/lib/node_modules/pm2/lib/API/Startup.js:305
      throw new Error('Unknown platform / init system name');
      ^

Error: Unknown platform / init system name
    at API.CLI.startup (/usr/lib/node_modules/pm2/lib/API/Startup.js:305:13)
    at Command.<anonymous> (/usr/lib/node_modules/pm2/bin/pm2:741:9)
    at Command.listener (/usr/lib/node_modules/pm2/node_modules/commander/index.js:315:8)
    at Command.emit (events.js:180:13)
    at Command.parseArgs (/usr/lib/node_modules/pm2/node_modules/commander/index.js:651:12)
    at Command.parse (/usr/lib/node_modules/pm2/node_modules/commander/index.js:474:21)
    at Timeout._onTimeout (/usr/lib/node_modules/pm2/bin/pm2:216:15)
    at ontimeout (timers.js:466:11)
    at tryOnTimeout (timers.js:304:5)
    at Timer.listOnTimeout (timers.js:267:5)
pi@mirror:~ $ pm2 startup
[PM2] Init System found: systemd
[PM2] To setup the Startup Script, copy/paste the following command:
sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
pi@mirror:~ $ sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
[PM2] Init System found: systemd
Platform systemd
Template
[Unit]
Description=PM2 process manager
Documentation=https://pm2.keymetrics.io/
After=network.target

[Service]
Type=forking
User=pi
LimitNOFILE=infinity
LimitNPROC=infinity
LimitCORE=infinity
Environment=PATH=/usr/bin:/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Environment=PM2_HOME=/home/pi/.pm2
PIDFile=/home/pi/.pm2/pm2.pid

ExecStart=/usr/lib/node_modules/pm2/bin/pm2 resurrect
ExecReload=/usr/lib/node_modules/pm2/bin/pm2 reload all
ExecStop=/usr/lib/node_modules/pm2/bin/pm2 kill

[Install]
WantedBy=multi-user.target

Target path
/etc/systemd/system/pm2-pi.service
Command list
[ 'systemctl enable pm2-pi' ]
[PM2] Writing init configuration in /etc/systemd/system/pm2-pi.service
[PM2] Making script booting at startup...
[PM2] [-] Executing: systemctl enable pm2-pi...
Created symlink /etc/systemd/system/multi-user.target.wants/pm2-pi.service → /etc/systemd/system/pm2-pi.service.
[PM2] [v] Command successfully executed.
+---------------------------------------+
[PM2] Freeze a process list on reboot via:
$ pm2 save

[PM2] Remove init script via:
$ pm2 unstartup systemd
pi@mirror:~ $ pm2 start ~/MagicMirror/installers/pm2_MagicMirror.json
[PM2] Spawning PM2 daemon with pm2_home=/home/pi/.pm2
[PM2] PM2 Successfully daemonized
```

MichMich, 2018-08-16:
This might need a change indeed. Feel free to send a PR. Thanks for reporting.

kosukekurimoto, 2018-08-17:
I solved this problem by modifying the installer script.

https://github.com/MichMich/MagicMirror/blob/6aa156d95667b230264fd99c5df877b25759e710/installers/raspberry.sh#L155

```
sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"
↓
sudo su -c "env PATH=$PATH:/usr/bin pm2 startup u pi --hp /home/pi"
```

ecosio1, 2018-09-07:
Im having a similar problem where it doesn't auto start on the monitor but for some reason when im in the VNC serer it shows it on that screen? Has anyone encountered this? (see far left(VNC) and far right screen(PI Monitor))
[image]
.

sdetweil, 2019-08-22:
i have a test update for pm2, both as part of install, and standalone 

standalone
```
https://forum.magicmirror.builders/topic/10907/new-script-to-install-fix-pm2-config-for-start-on-boot
```
updated installer, 
```
https://forum.magicmirror.builders/topic/10171/anyone-want-to-try-updated-installer
```
once tested more i will submit pr

closed 2019-10-28 by stale[bot]