MagicMirrorOrg/MagicMirror #1382: Expose option to select modules by name.

edward-shen, 2018-08-17:
This is a feature request.

**Platform**: Raspberry Pi

**Node Version**: 8.9.3
**MagicMirror Version**: Latest stable 2.4.1

**Description**: Currently, modules can't filter or target direct modules by name. Currently, this is not a problem, as `withClasses`, by default, provides the name of the module. This makes it easy for modules for MagicMirror modules such as MMM-pages to control other module's visibility.

However, this is dependent on having a module maintainer either appending to `withClasses` or leaving it untouched. If a maintainer changes this field, it prevents other modules from controlling visibility.

This issue first was raised by a user on edward-shen/MMM-Pages#8 and the maintainer of MMM-Weather-Now refused to implement a PR I made, believing that this should be resolved upstream. 

Please let us know if this is an intended feature, and if not, if you would appreciate a PR.

MichMich, 2018-08-17:
Not sure if I understand your question correctly, but in the current situation, [the name of the module will be added to the list of classes](https://github.com/MichMich/MagicMirror/blob/master/js/loader.js#L105). Because of this you can filter or target modules based on the name:

```
MM.getModules().withClass("clock")
```

Keep in mind that a configuration can include multiple instances of the same module.

MichMich, 2018-08-17:
Ah, I just checked the issue you mentioned on the MMM-Weather-Now module. Indeed, the classes property is intended to be used as read only, should at least adhere to the original behaviour.

Therefor this issue should be solved in the MMM-Weather-Now repository.

edward-shen, 2018-08-17:
Thanks for your quick response and clarification in the PR. I'll close this issue since this is not a problem relevant to the MagicMirror Core.

closed 2018-08-17 by edward-shen