MagicMirrorOrg/MagicMirror #1399: Why a Server and Client?

lacymorrow, 2018-09-03:
I know there's a reason I'm not thinking of, but why wouldn't it make sense to have the MMM-Server installed and running the display?

There would be no need for a separate Client.

Control, settings, startup would be through SSH.

I know someone smarter than myself has a good reason :)

ToddKlindt, 2018-09-03:
Multiple clients. I have several clients pointing to one Magic Mirror. When I make changes I only have to update the server and all of the clients get the changes.

lacymorrow, 2018-09-05:
@ToddKlindt Thanks for the quick reply, that definitely makes sense from a every functional standpoint except one: the "lazy guy" :)

I guess my question should be, how do most people have a single instance of a MM2 set up?

Do you run a server docker instance and use the Pi thread as the client? 

Apologies if this is somewhere obvious on the Wiki, I'm just trying to setup my first MM and the thought of needing a second box as a server running somewhere is a bit confusing to me.

(Definitely sounds like something excellent to try though! Screens through the house??)

lacymorrow, 2018-09-05:
feel free to comment your setup as well @ToddKlindt, I'm still getting ideas

ToddKlindt, 2018-09-05:
I run MM on a Pi 3 with a display attached to it. I have 2 other Pi Zero Ws hooked up to other TVs. I also hit the MM page from a browser on my desktop from time to time, like when I'm traveling.

lacymorrow, 2018-09-05:
@ToddKlindt, makes sense now. Love the setup and have a box of old pis that would make for great client boxes.

Would you happen to know if electron is running on the client, or rendered on the server and sent?

MichMich, 2018-09-05:
@lacymorrow Just FYI, the default install is an all-in-one installation. There is no need for a separate server and client in 99% of te cases. However, for those who want to separate the front and backend can use a server-only and client-only solution.

The Electron app is basically the browser showing the interface. So Electron is always running client side.

If you start your first MM project, I'd recommend going for the default installation. No need to complicate things.

closed 2018-09-06 by lacymorrow