MagicMirrorOrg/MagicMirror #1406: Ongoing events that started in past are not displayed

calli23, 2018-09-18:
**Platform**: Raspberry Pi 3b

**Node Version**: v9.11.2

**MagicMirror Version**: v2.

**Description**: Ongoing events that already started in past are not in the event list.
**Steps to Reproduce**: 
1. Create a calender series with a weekly frequence and events that lasts one week. The events should repeat e. g. every six week. 
2. Make sure that your event starts on the same weekday like yesterday, so that a the event started yesterday and lasts for seven days.
3. Integrate that calendar into your MagicMirror board.
4. The event starting from yesterday is not shown, but future events are shown.

**Expected Results**: I'd expect to see the events that started in past and are ongoing on MagicMirror calender module.

**Actual Results**: See point 4 in steps to reproduce.

**Configuration**: No special configuration. Just symbol an url is configured.

**Additional Notes**: 
I've created a hard coded workaround in calenderfetcher.js, that sets the startdate for between function to a value of last month. But that only works in my case and could be done much smarter. See my workaround code:
```

if (typeof event.rrule != "undefined" && !isFacebookBirthday) {
	var rule = event.rrule;	
	var lastMonth = moment().startOf("day").subtract(1, "months").toDate();
	var dates = rule.between(lastMonth, future, true, limitFunction);
        ...
```

lavolp3, 2018-11-21:
Oddly, this ONLY counts for recurring events that are currently ongoing...
If you set up a one-time event started yesterday and still ongoing, it does show up in the calendar.

Ultimatum22, 2019-05-30:
I was in a bugfix mood so I looked in this issue. I don't have a fix yet, but here is a breakdown of the issue.

All events are checked for start and end date, then there is a rule check (rule include any information about frequency of the event, daily, weekly, count etc).

During this rule check the problem occurs. When a repeating event is found all start and end dates of frequency are calculated. But to find which dates of the repeating event is needed, there is code to search from past to future, by default past is equal to the date of today. Which means repeating events that occur from today till future are shown.

My tracelog for claification
**Today is [Wednesday, May 29, 2019 4:00:00 AM]**
I made 4 events, but only the last 3 are found in the rule section because the first one is excluded based on startdate is before today date

> Event: Repeat
> Event: Repeat >> start 1559102400000 [Wednesday, May 29, 2019 4:00:00 AM]
> Event: Repeat >> duration 608400000
> Event: Repeat >> rule FREQ=WEEKLY;WKST=SU;COUNT=3;BYDAY=WE;DTSTART=20190529T040000Z
> Event: Repeat >> rule [startDate] 1559620800000 [Tuesday, June 4, 2019 4:00:00 AM]
> Event: Repeat >> rule [startDate] 1560225600000 [Tuesday, June 11, 2019 4:00:00 AM]
> Event: Repeat >> rule [startDate] 1560830400000 [Tuesday, June 18, 2019 4:00:00 AM]

There are config items to show items in the past but haven't got that working yet, when I specify broadcastPastEvents and maximumNumberOfDays no calendar events are shown anymore.

closed 2019-10-11 by stale[bot]