MagicMirrorOrg/MagicMirror #1414: problem with currentweather...

justjim1220, 2018-09-25:
it seems the link you placed to get the city id is no longer a working link

http://openweathermap.org/help/city_list.txt

Gives a ERROR 404 PAGE NOT FOUND

MichMich, 2018-09-25:
Thanks for the heads up. Where did you see that link?

el97, 2018-09-25:
> 
> 
> Thanks for the heads up. Where did you see that link?

The link is used for the default module [Weather forecast](https://github.com/MichMich/MagicMirror/tree/master/modules/default/weatherforecast)

justjim1220, 2018-09-25:
its in the Read Me

justjim1220, 2018-09-25:
further investigation has this on the OpenWeather website:

By city ID
Description:
You can call by city ID. API responds with exact result.

List of city ID city.list.json.gz can be downloaded here http://bulk.openweathermap.org/sample/

We recommend to call API by city ID to get unambiguous result for your city.

Parameters:
id City ID
Examples of API calls:
api.openweathermap.org/data/2.5/weather?id=2172797

justjim1220, 2018-09-25:
I found the json file...
Converted it to text...

But, it's a 30MB file, too big to upload here...

justjim1220, 2018-09-25:
here is the compressed file....
Can extract it with winwar, 7-zip, et. al.

[city.list.json.gz](https://github.com/MichMich/MagicMirror/files/2416674/city.list.json.gz)