MagicMirrorOrg/MagicMirror #1416: create config file

LeyDi1987, 2018-09-28:
Hi,
My Magicmirror isnt working anymore. Says it needs to create a config file.
The only thing that changed today is my WIFI connection. but my raspberry is good connected. What is wrong?

this is my config

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
	                                                       // or add a specific IPv4 of 192.168.1.5 :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
	                                                       // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
	                                                       // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	language: "en",
	timeFormat: 24,
	units: "metric",

	modules: [
		
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "Agenda",
			position: "top_left",
			config: {
				calendars: [
					{dateFormat:'absolute' 
						symbol: "calendar-check-o ",
						url: "webcal://p13-calendars.icloud.com/published/2/EfsP4Zi_PkxuRIMRT0p8bKj_d0gyV4k2de-R7UDiLJHsIRXFN7Ohx97sjMBd_PQAgckxZFkQYXbf6HH2u678tG-ok7TY8_rL7kp2_MBrPsU"
					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Maastricht",
				locationID: "2751283",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "361d187b266e92a8b8c0cf49f318099d"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "Maastricht",
				locationID: "2751283",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "361d187b266e92a8b8c0cf49f318099d"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Nos Nieuws",
						url: "http://feeds.nos.nl/nosnieuwsalgemeen"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
                {
	module: 'MMM-Traffic',
	position: 'bottom_left',
	classes: 'bright medium', //optional, default is 'bright medium', only applies to commute info not route_name
	config: {
		api_key: 'ushed my api key',
		mode: 'driving',
		origin: 'Aubeldomein 14, Maastricht',
		destination: 'Spoorweghaven 323, Rotterdam',
		mon_destination: '',
		fri_destination: '',
		arrival_time: '', //optional, but needs to be in 24 hour time if used.
		route_name: 'Huis naar werk',
		changeColor: true,
		showGreen: true,
		limitYellow: 5, //Greater than 5% of journey time due to traffic
		limitRed: 20, //Greater than 20% of journey time due to traffic
		traffic_model: 'best_guess',
		interval: 120000 //2 minutes
		}
	},

	
       
},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

el97, 2018-09-28:
It looks like you missed some commas and stuff... I fixed it for you 👍 https://pastebin.com/N6XVWQpq

MichMich, 2018-10-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-10-01 by MichMich

justjim1220, 2018-10-01:
you are missing a comma after `{dateFormat:'absolute' `
should be like this...
`
calendars: [
	{
            dateFormat:'absolute' ,
	    symbol: "calendar-check-o ",
`