MagicMirrorOrg/MagicMirror #1420: Blank page with new version...

justjim1220, 2018-10-01:
here's my errors from Developer tools...

[image]

justjim1220, 2018-10-01:
I did the npm start dev, but can't find the log file to post for you...

MichMich, 2018-10-01:
Can you try restarting your pi?

justjim1220, 2018-10-01:
I am not using a pi. I use Windows on PC's.

I replaced the defaults folder with the backup I made. Now it works fine.

I will try again with the new defaults directory and reboot my PC.

justjim1220, 2018-10-01:
Thank You!

Restarting Worked!

I guess I need to do that more often... The third time recently that restarting my PC helped get a module to work!

Sorry to bother you!

MichMich, 2018-10-01:
No worries! Glad it's solved.

closed 2018-10-01 by MichMich