MagicMirrorOrg/MagicMirror #1423: Calendar issues

GGRIZLI, 2018-10-03:
R-Pi 3b, win10, chrome
MagicMirror laterst version (2.5.0)

When I update MagicMirror to the latest version, date on calendar change to 2 days, like:

**Dan suverenosti -> 25.okt. - 26.okt.**

must be:

**Dan suverenosti -> 25.okt.**

and this is some code of my config file:

> module: "calendar",
			header: "GRIZLIJEV KOLEDAR",
			position: "top_left",
			config: {
				colored: true,
				maximumEntries: 8,
				timeFormat: "absolute",
				dateFormat: "Do MMM",
				dateEndFormat: "HH:mm",
				fullDayEventDateFormat: "Do MMM",
				urgency: 0,
				getRelative: 6,
				fadePoint: 0.30,
				maxTitleLength: 30,
				wrapEvents: true,
				calendars: [
					{
						symbol: "gift",
						url: "https://calendar.google.com/calendar/ical/....

> 

Where is the catch?

henrikra, 2018-10-26:
I think the reason for you is also that you need to set `showEnd` to `false` in calendar module's config. Check my issue also which is related to this: https://github.com/MichMich/MagicMirror/issues/1450

GGRIZLI, 2018-10-26:
Yes, thats it...  tnx.

Have i nice day 😁

closed 2018-10-26 by MichMich