MagicMirrorOrg/MagicMirror #1435: full day events show next day as EndDate

Klami, 2018-10-07:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

AndyHazz, 2018-10-09:
Could this be an unwanted side effect of this change in 2.5.0?
- Consider multi days event as full day events

Any full day events (single day or multi-day) from my google calendar now look confusing when displayed ... they always display the day after the last day of the event as the end date. e.g. if I have a full day event across Saturday/Sunday, it will incorrectly display in MagicMirror as ending on Monday.

iammurdock92, 2018-10-16:
Also having this issue

bacquy, 2018-10-18:
Same problem here.

AndyHazz, 2018-10-30:
The changes for #1450 are good if you don't ever want to use this feature, but I think the feature still needs work for anyone that is interested in seeing EndDate?

Ideally, EndDate should show the last day the event takes place on, not the day after when the event is no longer happening. It should also only show if different to the start date.

Otherwise, Birthdays always last 2 days (I suppose that could be a good thing), and things like holidays all appear to last a day longer than planned (again, that sounds quite good, maybe we should leave it).