MagicMirrorOrg/MagicMirror #1436: Weathreforecast - Rain Amount not working - API changes?

Fredde87, 2018-10-07:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).
Raspberry Pi 3, electron 3.0.3

**Node Version**: Make sure it's version 0.12.13 or later.
9.11.2

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).
2.4.0

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

Rainfall is never shown as forecast.rain is always NaN.

API appears different from what the weatherforecast.js file expects when it processes the rain.
The javascript appears to just check for a forecast.rain value. However running the API request manually and/or reading the Openweather API documentation reveals that results are returned in mm for each 3 hour period (forecast.rain.3h etc).
For rain amount to be correctly processed, the weatherforecast.js file needs to calculate and add all the rainfall together for that day to show a 24h total instead of a 3h total.
There appears to be no such code in place, so I am guessing the API has recently changed from Openweather's side?

**Steps to Reproduce**: List the step by step process to reproduce the issue.
In config.js, add line to weatherforecast module,
showRainAmount: true

**Expected Results**: Describe what you expected to see.
Rainfall in mm

**Actual Results**: Describe what you actually saw.
No rainfall (forecast.rain isNaN)

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
showRainAmount: true

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Fredde87, 2018-10-08:
Ahh that does make sense. The thought did cross my mind for a split second but then forgot about it. As I dont have a premium account I had no way to test it myself.

MichMich, 2018-10-30:
Can I close this issue?

Fredde87, 2018-11-01:
> Can I close this issue?

I dont think so because we need to either,

A) Update the Wiki to explain this feature is only available with paid premium accounts
or
B) Add support for the free accounts as well

Option B is obviously the best route but Option A is technically valid too.

Fredde87, 2018-12-20:
> We can't add option B. Option A is something you can simply contribute yourself. :)

Sorry, do you mean there is a technical reason option B can't be done? Or do you mean there is just not enough demand for it?

Just asking because it is on my list of things to look at doing if I get some spare time after the new year. But if there is a technical reason it can't be done then I can just do option A straight away :)

MichMich, 2018-12-20:
OpenWeatherApp stopped supporting it for free accounts, so that's out of our hands. If you want to spend time on it, you might want to dive into the weather-refactor branche which has a new "Weather" module. This module should allow us to add new (different) weather providers mode easily. Eliminating the need for (paid) openweathermap accounts.

See: https://github.com/MichMich/MagicMirror/tree/weather-refactor/modules/default/weather

Fredde87, 2018-12-20:
> OpenWeatherApp stopped supporting it for free accounts, so that's out of our hands. If you want to spend time on it, you might want to dive into the weather-refactor branche which has a new [...]

So the current weather module will be replaced by this one?

You sure they stopped the support? When I created this github issue then the problem was not that they dont do it for free accounts. It is just they report it back in 3h intervals instead of a 24h interval so we just need a loop to go through each 3h segment and add it all up.

MichMich, 2018-12-20:
Yeah, that one would replace the current weather modules. Not only because OpenWeatherMap stopped full support for free account but also because their forecasts are pretty flacky for some locations. 

Feel free to send PRs for the new Weather module or the existing weather modules.

fewieden, 2018-12-22:
I'm planning to address the weather refactoring in the hackathon https://forum.magicmirror.builders/topic/9298/magicmirror-hackathon-2018

https://github.com/fewieden/MagicMirror-Hackathon/issues/12

vincep5, 2019-05-28:
This can be closed.  In the new weather module, the code handles forecast and forecast/daily

https://github.com/MichMich/MagicMirror/blob/b508a629e8a727358f1b88e3c7df842d14669ec4/modules/default/weather/providers/openweathermap.js#L153-L159

https://github.com/MichMich/MagicMirror/blob/b508a629e8a727358f1b88e3c7df842d14669ec4/modules/default/weather/providers/openweathermap.js#L201-L207

closed 2019-05-29 by MichMich