MagicMirrorOrg/MagicMirror #1438: Sudden Black Screen Today, Cannot Fix

chiefanand, 2018-10-08:
As of this morning I suddenly began to have a black screen instead of any modules showing. Reboot and full MM2 install does not change anything.

pi@raspberrypi:~ $ pm2 start mm.sh
[PM2] Applying action restartProcessId on app [mm](ids: 0)
[PM2] [mm](0) ✓
[PM2] Process successfully started
┌──────┬──────┬────────┬───┬─────┬──────────┐
│ Name │ mode │ status │ ↺ │ cpu │ memory   │
├──────┼──────┼────────┼───┼─────┼──────────┤
│ mm   │ fork │ online │ 0 │ 0%  │ 2.6 MB   │
└──────┴──────┴────────┴───┴─────┴──────────┘
 Use `pm2 show <id|name>` to get more details about an app
pi@raspberrypi:~ $ pm2 logs mm
[TAILING] Tailing last 15 lines for [mm] process (change the value with --lines option)
/home/pi/.pm2/logs/mm-error.log last 15 lines:
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | libGL error: MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | libGL error: MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | [858:1008/121651.676339:FATAL:bus.cc(1152)] D-Bus connection was disconnected. Aborting.
0|mm       | libGL error: MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | libGL error: MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information

/home/pi/.pm2/logs/mm-out.log last 15 lines:
0|mm       | ====================== Starting node_helper for module [MMM-MyCommute]
0|mm       | Connecting socket for: newsfeed
0|mm       | Starting module: newsfeed
0|mm       | Sockets connected & modules started ...
0|mm       | Launching application.
0|mm       | Shutting down server...
0|mm       | Stopping module helper: calendar
0|mm       | Stopping module helper: MMM-RottenTomatoes
0|mm       | Stopping module helper: MMM-3Day-Forecast
0|mm       | Stopping module helper: MMM-MyCommute
0|mm       | Stopping module helper: newsfeed
0|mm       | 
0|mm       | > magicmirror@2.5.0 start /home/pi/MagicMirror
0|mm       | > sh run-start.sh
0|mm       | 

0|mm  | Starting MagicMirror: v2.5.0
0|mm  | Loading config ...
0|mm  | Loading module helpers ...
0|mm  | No helper found for module: alert.
0|mm  | No helper found for module: clock.
0|mm  | Initializing new module helper ...
0|mm  | Module helper loaded: calendar
0|mm  | No helper found for module: compliments.
0|mm  | Initializing new module helper ...
0|mm  | Module helper loaded: MMM-RottenTomatoes
0|mm  | Initializing new module helper ...
0|mm  | Module helper loaded: MMM-3Day-Forecast
0|mm  | Initializing new module helper ...
0|mm  | Module helper loaded: MMM-MyCommute
0|mm  | Initializing new module helper ...
0|mm  | Module helper loaded: newsfeed
0|mm  | All module helpers loaded.
0|mm  | Starting server on port 8080 ... 
0|mm  | Server started ...
0|mm  | Connecting socket for: calendar
0|mm  | Starting node helper for: calendar
0|mm  | Connecting socket for: MMM-RottenTomatoes
0|mm  | Connecting socket for: MMM-3Day-Forecast
0|mm  | MMM-3Day-Forecast helper, started...
0|mm  | Connecting socket for: MMM-MyCommute
0|mm  | ====================== Starting node_helper for module [MMM-MyCommute]
0|mm  | Connecting socket for: newsfeed
0|mm  | Starting module: newsfeed
0|mm  | Sockets connected & modules started ...
0|mm  | Launching application.
0|mm  | libGL error: MESA-LOADER: failed to retrieve device information
0|mm  | MESA-LOADER: failed to retrieve device information
0|mm  | MESA-LOADER: failed to retrieve device information

I tried all the troubleshooting guides, but I can't seem to fix it. I even tried reverting back to the stock config.js, but no luck. 

**Platform**: Raspberry Pi 3B+, Electron version 2.0.0

**MagicMirror Version**: Latest, 2.5

**Configuration**: Happy to provide any details, but even the stock, unaltered config.js appears as a black screen.

illskool, 2018-10-08:
I am having the same issue.  I upgraded to version 2.5 and my mirror displays black screen ever since.  Please help

chiefanand, 2018-10-12:
I ended up restoring a backup that was on the previous version of MM2 and removing the update alert module - running perfectly again! Not sure what the issue is at all, but it's working well enough for my needs now.

newbee46, 2018-10-15:
I created a new config file and placed in the configuration folder but still nothing shows up except a black screen.  Anyone have any ideas!

wkra, 2018-10-20:
12 days from bug report and no response or idea? Is it a marginal problem? Should I remove my MM and setup it from beginning or is there less invasive fix?

Joshua2504, 2018-10-27:
Mostly this in my chrome console.

`Refused to execute script from '<URL>' because its MIME type ('text/html') is not executable, and strict MIME type checking is enabled.`

`Refused to execute script from 'http://192.168.1.2:11061/vendor/node_modules/moment/min/moment-with-locales.js' because its MIME type ('text/html') is not executable, and strict MIME type checking is enabled.`

jo-cee, 2018-10-27:
Had the same problem, after copying parts of the config to the sample i could narrow it down to `address: ""`. If left blank MagicMirror showed only a black screen which could not be quit by ctrl+q. With localhost or 127.0.0.1 everything worked like expected. Additional, while the pi has only shown a black screen, Firefox on my PC  did show nearly a full Mirror when connected to the pi's IP, only the modules currentweather and waetherforecast did not show up.

Joshua2504, 2018-10-27:
> Had the same problem, after copying parts of the config to the sample i could narrow it down to `address: ""`. If left blank MagicMirror showed only a black screen which could not be quit by ctrl+q. With localhost or 127.0.0.1 everything worked like expected. Additional, while the pi has only shown a black screen, Firefox on my PC did show nearly a full Mirror when connected to the pi's IP, only the modules currentweather and waetherforecast did not show up. Edit: `address: "0.0.0.0"` works also.

Please check your Firefox console for any errors if you haven't done this already.

jo-cee, 2018-10-27:
```
Loading failed for the <script> with source “http://192.168.10.10:8080/modules/MMM-WatchDog//MMM-WatchDog.js”. 192.168.10.10:8080:1:1
Error on loading script: modules/MMM-WatchDog//MMM-WatchDog.js loader.js:190:5
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://api.openweathermap.org/data/2.5/weather?id=2867838&units=metric&lang=de&APPID=xxx. (Reason: CORS request did not succeed).[Learn More]
currentweather: Could not load weather. currentweather.js:330:6
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://api.openweathermap.org/data/2.5/forecast/daily?id=2867838&units=metric&lang=de&APPID=xxx. (Reason: CORS request did not succeed).[Learn More]
weatherforecast: Could not load weather. weatherforecast.js:268:6
```
(i deleted the APPID)

These errors show up in the console of Firefox when going back to the blackscreen-config. Disabling watchdog, same result, pi black, remote firefox missing weather.

MichMich, 2018-10-30:
Just that I understand correcty: is this issue fixed with the correct `address`?

jo-cee, 2018-10-30:
The 7 inch touch screen of my pi works normally when i use `localhost` or `127.0.0.1 `or `0.0.0.0`, these are the values i tested. It is black, when i left the adress `empty `or used `::`.

MichMich, 2018-11-06:
Since the problem is solved when you fill in the correct address: can I close this issue?

jo-cee, 2018-11-06:
Seems the OP did found a solution for his problem, my problem is fixed.

closed 2018-11-08 by MichMich

mistermj, 2018-11-08:
I am facing the same issue. Can you please let me know why does everything works for the first time but after a reboot, a empty black screen shows up?

mistermj, 2018-11-08:
And I always do Automatic Installation, shall I go with manual installation?

mistermj, 2018-11-08:
Worked perfectly fine with manual installation. Thanks.