MagicMirrorOrg/MagicMirror #1439: document / fix showEnd

tkoeberl, 2018-10-09:
Hi,

since the latest update the Dates are shown at the end of the calendar entry.
I tried to disable it via the config with showEnd: "false", or showEnd: false but didn't work out.

as a workaround I set the default to false in file: ./modules/default/calendar/calendar.js

regards,
-thomas

Duske, 2018-10-20:
I guess this should be fixed with #1437

BennyS91, 2018-10-23:
I had the same problem, I just added it direct in config.js as `showEnd: false,` as base configuration element in my calendar. This applies for all then, and now it looks fine again ;-).

Here for example:

```      {
            module: "calendar",
            header: "Kalender",
            position: "top_left",
            config: {
                colored: true,
                maximumEntries: 11,
                maxTitleLength: 40,
                showEnd: false,
                calendars: [{
                        symbol: "smile-o",
                       .........
```

tkoeberl, 2018-10-23:
Cool. thats it.

closed 2018-10-23 by tkoeberl