MagicMirrorOrg/MagicMirror #1454: After update 2 dates

Mattie38wit, 2018-11-05:
I have a small problem with my calender.
afterI updated my Magic Mirror. 2 dates appeard for one day.

like: Sinterklaasavond 5 Dec - 6 Dec.

this is because its a full day?, but can I change it to only 0 date

MrGrape188, 2018-11-06:
I'm experiencing the same problem here for full day events. Besides that I also noticed my calendar items with fixed times are 2 hours off, for instance displaying 14:00 instead of 12:00. Guess this is why full day events are spanning 2 days now.

I'm using Google calendar (iCal format). My timezone settings are both correct in Google Calendar and on my RaspberryPi ((GMT+1:00) Central European time - Amsterdam). Before updating it used to work without any problems.

MichMich, 2018-11-06:
See: https://github.com/MichMich/MagicMirror/issues/1450

closed 2018-11-06 by MichMich

MrGrape188, 2018-11-06:
Thanx, MichMich, for your prompt reply! I never would have figured out it was displaying the end date, just thought the start was a few hours off, but reading it now it makes sense!! All the best, Peter