MagicMirrorOrg/MagicMirror #1460: config.js error

swvalenti194, 2018-11-08:
Sample config.js works fine, when I change to below I get please create a config file. Only edited the bold at the moment would like to edit the news and weather but one step at a time.   Any help would be greatly appreciated

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	**address: "0.0.0.0",
	port: 8080
	ipWhitelist: {},
	language: "en",
	timeFormat: 24,
	units: "imperial",**

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check-o ",
						url: "webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics"
					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "",  //ID from http://bulk.openweathermap.org/sample/; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

swvalenti194, 2018-11-08:
i truly did my due diligence and didnt want to post such a newbie issue but im at a loss.  Ive done other builds with no issues idk what im doing wrong here.  Im on day 3 trying to fix config file to no avail

vrhax, 2018-11-09:
Your ipWhitelist needs to be ipWhitelist: [], instead of ipWhitelist: {},

swvalenti194, 2018-11-09:
I changed that and still get the error....is it because I am using the GUI to navigate to the folder of magic mirror and not the terminal?

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "0.0.0.0",
	port: 8080
	ipWhitelist: [],
	language: "en",
	timeFormat: 24,
	units: "imperial",

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check-o ",
						url: "webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics"
					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "",  //ID from http://bulk.openweathermap.org/sample/; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581",  //ID from http://www.openweathermap.org/help/city_list.txt
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

vrhax, 2018-11-09:
Try verifying your config using jslint: https://www.jslint.com/

I run into this prob on a regular basis, mostly due to missing commas but also due to node.js strict indentation policy. Also, if you edited your file on say, a windoze machine, you'll run into problems, as their carriage returns differ from linux machines.

At this point, the best thing to do is to ssh to your PI, and edit it there. In fact, since you've only made minimal modifications (i.e., time format, iplist, etc), just grab the sample config from the repository (link to raw code: https://raw.githubusercontent.com/MichMich/MagicMirror/master/config/config.js.sample), and cut/paste that into your config.

Then, do pm2 restart mm, and see if it works. If it does, edit your config file, and make one change, restart to ensure you haven't introduced any errors.

Rinse, repeat, for each change thereafter.

swvalenti194, 2018-11-10:
Ok I will give that a try but where do I copy and paste the config file to using commands from terminal?

vrhax, 2018-11-10:
Haven't a clue. I just ssh to my pi. Btw, none of this appears to actually be a bug. I would urge you to check out their google group general discussion:

https://groups.google.com/forum/#!forum/diy-magic-mirror

lavolp3, 2018-11-13:
The forum you should go to is rather [forum.magicmirror.builders](forum.magicmirror.builders).
That's the right place for these discussions.
For your remote control issue, you will find a solution there as well. Otherwise post it there again.