MagicMirrorOrg/MagicMirror #1469: [question] calendar, filter on category

mdissel, 2018-11-23:
Is it possible to filter events based  on one or more categories?

For example, tag your calendar entry with the category family, to show it on the family board

MichMich, 2018-11-29:
Currently not but feel free to send a PR.

lavolp3, 2018-11-30:
What you could do is to create a google sub-calender called family with the entries and include it in the family board.
Or name every entry something with "Family: " and have MM filter them out (there are filter rules implemented into the code)
However, you would need to do something with the events in the original calendar, don't you? 
I can't see away to "tag" inside the MagicMirror framework.

mdissel, 2018-11-30:
> What you could do is to create a google sub-calender called family with the entries and include it in the family board.

But then you have to re-enter family appointments again..

> However, you would need to do something with the events in the original calendar, don't you?

In my business calendar I want to tag events that should be visible on the 'family calender'/magic mirror.

closed 2019-10-11 by stale[bot]