MagicMirrorOrg/MagicMirror #1470: Newsfeed module out of date link

Charnub, 2018-11-24:
**Platform**: Raspberry Pi 3

**Node Version**: v9.11.2

**MagicMirror Version**: JavaScript version (v2)

**Description**: In the newsfeed module the link for BBC is out of date and is showing news from 8 months ago to 2 years ago.

**Steps to Reproduce**: By placing the following in `config.js`: 
> {
					title: "BBC",
					url: "http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=uk",
				},

**Expected Results**: Expected to see up to date news with current results.

**Actual Results**: Greeted with very old news.

**Configuration**: 

> {
                        module: "newsfeed",
                        position: "bottom_bar",
                        config: {
                                feeds: [
                                        {
                                                title: "BBC News",
                                                url: "http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=uk",
                                        }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },

**Additional Notes**: A fix is available however it is dependent on where you would like news from. I am using: 

> url: "http://feeds.bbci.co.uk/news/world/europe/rss.xml",

However if you go to [https://www.bbc.co.uk/news/10628494#userss](url) then look at the links under Global and UK News Feeds then copy the respectable link then you will get up to date news.

MichMich, 2018-11-29:
It seems you are just using an old link in your config. this is not something in the framework defaults, is it?

MichMich, 2018-12-19:
Closed due to inactivity.

closed 2018-12-19 by MichMich