MagicMirrorOrg/MagicMirror #1471: Uncaught Exception preventing METAR data from displaying

martyg33, 2018-11-28:
Platform: Raspberry Pi 3

Node Version: node 2.5.0

MagicMirror Version: v2-beta (unsure)

Description:: Unable to load METAR data, even though it has been downloaded. Mirror starts and runs fine, but the Aviation module simply does not populate. This is the error I am getting:

FAA data found for KMSP
Whoops! There was an uncaught exception...
SyntaxError: Unexpected token < in JSON at position 0
    at JSON.parse (<anonymous>)
    at Request._callback (/home/pi/MagicMirror/modules/MMM-aviationwx/node_helper.js:60:48)
    at Request.self.callback (/home/pi/MagicMirror/modules/node_modules/request/request.js:185:22)
    at emitTwo (events.js:126:13)
    at Request.emit (events.js:214:7)
    at Request.<anonymous> (/home/pi/MagicMirror/modules/node_modules/request/request.js:1161:10)
    at emitOne (events.js:116:13)
    at Request.emit (events.js:211:7)
    at IncomingMessage.<anonymous> (/home/pi/MagicMirror/modules/node_modules/request/request.js:1083:12)
    at Object.onceWrapper (events.js:313:30)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

Any help or direction you can provide is greatly appreciated.

martyg33, 2018-11-28:
Node Version is v9.11.1

MichMich, 2018-11-29:
Please open a issue in the MMM-aviationwx repository. Thanks.

closed 2018-11-29 by MichMich