MagicMirrorOrg/MagicMirror #1472: [question] error??

igroo2010, 2018-11-29:
Platform: Raspberry Pi 3

Node Version: v9.11.2

MagicMirror Version: JavaScript version (v2)

Description: This is an error that occurs when Watchdog is executed.

Steps to Reproduce: Mirror start

Whoops! 
There was an uncaught exception...
{ Error: listen EADDRINUSE 127.0.0.1:8080
    
at Object._errnoException (util.js:1024:11)
    
at _exceptionWithHostPort (util.js:1046:20)
    
at Server.setupListenHandle [as _listen2] (net.js:1351:14)
    
at listenInCluster (net.js:1392:12)
    
at GetAddrInfoReqWrap.doListen [as callback] (net.js:1501:7)
    
at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:97:10)
  
code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080 }

What's the reason?

MichMich, 2018-11-29:
Port 8080 is already in use. 

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2018-11-29 by MichMich

igroo2010, 2018-11-30:
Thank you os much
 
Closed #1472.