MagicMirrorOrg/MagicMirror #1473: LXDE isn't installed in the last version of raspbian stretch

ishepard, 2018-11-30:
As explained in this answer (https://raspberrypi.stackexchange.com/a/91371), the new version of raspbian `2018-11-13-raspbian-stretch.img` doesn't contain LXDE. This implies that the `autostart` file doesn't exist anymore (or at least not in `~/.config/lxsession/` since it doesn't exist), and so we can not modify the orientation of the screen, hide the mouse pointer, or disabling the screensaver.

Now, for the first point (orientation of the screen), the solution I found was to put this line in `/etc/lightdm/lightdm.conf`:

```
[SeatDefaults]
display-setup-script=xrandr --output HDMI-1 --rotate [left or right]
```

Do you have any ideas on how to solve the other issues? 

**Platform**: Raspberry Pi 3. (but potentially affects all Raspberry Pi)

**Node Version**: v10

**MagicMirror Version**: v2

Gilaadb, 2018-12-06:
I'm using the same setup as you do. I used the auto install script but for some reason the autostart didn't function out of the box.
I looked at the [automated installer script](https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh), and just repeated the last section that is responsible for the autostart and it worked perfectly:
`sudo npm install -g pm2`
`sudo su -c "env PATH=$PATH:/usr/bin pm2 startup linux -u pi --hp /home/pi"`
`pm2 start ~/MagicMirror/installers/pm2_MagicMirror.json`
`pm2 save`

Hope this helps.

ishepard, 2018-12-07:
yep that can happen.
However, in my issue, I wasn't referring to a problem on _installing_ MagicMirror. That I have done it.
The problem is the **Configuration**, like removing the blank screen, mouse pointer, etc. because in the tutorial everything refers to the "autostart" file, which doesn't exist anymore.

Gilaadb, 2018-12-07:
I followed the tutorial and created the files they refer to.
The blank screen is behind my mirror window. That is, if i click F11, I see a blank (black) screen. But since it's in the background, I don't think it's a problem.
the mouse pointer is automatically hidden after the period of time I stated in the autostart file. I tried to increase the delay and could see it's effect.

lavolp3, 2018-12-09:
As far as I can see the latest Raspbian Stretch still includes LXDE.

> Raspbian is now spun in a minimal install image as well as a "Raspbian Full" image depending upon your preference for how many extra applications are bundled with the OS image. The minimal image still includes their LXDE-based desktop and some applications like Chromium and VLC

Changelog for 2018-11-13 says:

>   * Modifications to LXDE components to enable local configuration to override global configuration correctly

Are you sure you have not installed the light version? Sorry for the possibly insulting question :-)

Fredde87, 2018-12-20:
I think I have had the same thing today. I installed Raspberian Full Desktop via Noobs v3 last night on a brand new Pi3b+. It booted up with some sort of desktop environment but I didn't pay attention to which one it was, just installed SSH and went about setting up the Pi.
Installed various things including MagicMirror (did notice I had no lxsessions folder) and rebooted. Now just get a OpenBox black screen where I can launch a browser or terminal window from.
Had a look at apt and to my surprise lxde is not installed. So not sure if it somehow got uninstalled or if I was using a different environment before (Pixel?).
Also not sure what step made me loose my desktop but will look into it after Christmas

malikmalikmalik, 2018-12-28:
I fixed autostarting with pm2 by executing the following:
`sudo pm2 startup systemd -u pi --hp /home/pi`

Telling pm2 to start using systemd with user pi (if that's your user).
But, still stuck on the other topics.

There is no  ~/.config/lxsession/ folder.
I installed Raspbian Stretch with desktop today from the Raspberry Pi website.

Anyone got tips here?
This is the only lx folder I can find in .config/ :
 ~/.config/**lxpanel**/LXDE-pi/panels/

Maerchensarg, 2018-12-29:
I cant even start MM with the latest Raspbian Image file and dtoverlay=vc4-kms-v3d: 

libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information

Crashes finally at: Launching application.

---

And there is no ~/.config/lxsession/ folder on my new, freshly RPi3B+ Raspbian setup.

At the current state I can only use MM in landscape mode with the option to overheat the RPi.

Maerchensarg, 2018-12-29:
If I put dtoverlay=vc4-kms-v3d or dtoverlay=vc4-fkms-v3d in the /boot/config.txt, then the MM startup process stops at "Launching application" and I can only move the mouse. Every else is freezed. If I remove the entry MM starts flawless.

After creating the file "autostart" inside of "~/.config/lxsession/LXDE-pi/" with the content "@xrandr --output HDMI-1 --rotate right" I end up with a black screen, but working GUI. Seems to work, because it is rotated and I have a right click menu where I can start the terminal.

Fredde87, 2018-12-29:
> If I put dtoverlay=vc4-kms-v3d or dtoverlay=vc4-fkms-v3d in the /boot/config.txt, then the MM startup process stops at "Launching application" and I can only move the mouse. Every else is freezed. [...]

Yeah you seem to have the exact same issue I had as well. You just get a black OpenBox window manager right? You can launch a web browser and a terminal by right clicking the background?

I haven't tried it yet, but I think the reason you dont get LXDE launching is because the autostart file for it is missing. I have read that if you copy /etc/xdg/lxsession/LXDE/autostart to ~/.config/lxsession/LXDE/autostart then this should start LXDE when you reboot your Raspberry Pi.

What happens if you open a terminal and run pm2 start mm?

lavolp3, 2018-12-29:
I couldn't load the mirror with OpenGL driver as well. In my case the problem originated in the MMM-PIR-Senor module.
You can check that by checking
`pm2 logs`
via ssh when you have the black screen and check the error logs for module names. 
This module compiles some program to a certain environment. If the eonivronment changes in some aspect, the module breaks the mirror (that's my interpretation).

Maerchensarg, 2018-12-29:
@Fredde87, yes, exactly the same behavoir. I tried now:

pi@MAGICMIRROR:~ $ cat ~/.config/lxsession/LXDE/autostart
@lxpanel --profile LXDE
@pcmanfm --desktop --profile LXDE
@xscreensaver -no-splash

pi@MAGICMIRROR:~ $ cat ~/.config/lxsession/LXDE-pi/autostart
@xrandr --output HDMI-1 --rotate right
pi@MAGICMIRROR:~ $ 

The result is the same. Black screen with right click menu. "pm2 start mm" is not working for me, so I followed this guide to get it working (in the terminal emulator via right click):

https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror

After starting MM with "pm2 start mm.sh" pm2 tolds me everything is fine and a few seconds later MM pops up.

Current state: 

pi@MAGICMIRROR:~ $ cat /boot/config.txt 
[...]
dtoverlay=vc4-kms-v3d

Both LXDE configs from above and working "pm2 start mm.sh" start script. It is only working by hand. When I push a "pm2 save" and reboot the RPi3B+, it will end with that black screen again. From here I can start MM again via terminal and "pm2 start mm.sh".

pi@MAGICMIRROR:~ $ pm2 save
[PM2] Saving current process list...
[PM2] Successfully saved in /home/pi/.pm2/dump.pm2

REBOOT

pi@MAGICMIRROR:~ $ pm2 show mm
[PM2][WARN] mm doesn't exist

ishepard, 2018-12-29:
So, I solved the problem of not having the `autostart` file by adding these lines in `/etc/lightdm/lightdm.conf`:

```
[SeatDefaults]
display-setup-script=xrandr --output HDMI-1 --rotate right
xserver-command=X -s 0 -dpms
```

The first one is for rotating the screen on the right, the second one is for deleting the black screen. So no need for the `autostart` file anymore.

Maerchensarg, 2018-12-29:
> So, I solved the problem of not having the `autostart` file by adding these lines in `/etc/lightdm/lightdm.conf`:
>
> ```
> [SeatDefaults]
> display-setup-script=xrandr --output HDMI-1 --rotate right [...]

I can confirm this. I changed the lines inside of the [Seat:*] block. Thanks!

ishepard, 2018-12-29:
@MichMich not sure how you want to address this issue of not having the `autostart` anymore, but the solution that seems to work is to put these 2 lines on [SeatsDefault] in `/etc/lightdm/lightdm.conf`:

```
[SeatDefaults]
display-setup-script=xrandr --output HDMI-1 --rotate right
xserver-command=X -s 0 -dpms
```

First one to rotate the display, second to delete black screen. The good thing is that the tutorial page on how to configure the raspberry will be much shorter :) 
Everything else that involves the `autostart` doesn't work (creating the file manually, etc.). Everything leads to unexpected behaviour, like freezing and `pm2` not able to start.

jerbookins, 2019-02-05:
I tried manually creating the directory for the autostart file as described in the configuration instructions:

`~/.config/lxsession/LXDE-pi/autostart`

and when booted it worked by rotating the screen but only showed a black screen and I couldn't figure out how to get anything else to work. Deleted the directory to restore GUI function.

As an aside, is it normal after running the autoinstaller and rebooting for the GUI to be pretty neutered? (e.g. no taskbar, smaller interface elements, no "start menu", etc?)

Fredde87, 2019-02-06:
> I tried manually creating the directory for the autostart file as described in the configuration instructions:
>
> `~/.config/lxsession/LXDE-pi/autostart`
>
> and when booted it worked by rotating the [...]

So I just reinstalled my MagicMirror today because I bought a SSD for my Raspberry Pi. So I ran into all these issues again so I did a bit of research.

The reason you loose all your menu bars etc is because if you create a autostart file for an individual user (pi), then it no longer reads the global autostart file (located under /etc/xdg/lxsession/LXDE-pi/autostart).

So your options are to either
A) Copy the content from the global autostart into your /home/pi/.config/lxsession/LXDE-pi/autostart or,
B) Put all your commands in the global file and don't create a user autostart file.

If you examine this global autostart you'll see it contains the following which starts lxde,

@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
point-rpi

This is what is no longer run if you create a user autostart file without copying the above lines into it.

Second in regards to @Maerchensarg's issue with MagicMirror saying Launching Application and then nothing happens (Electron starts but draws nothing so it appears that the Pi locks up but you can Alt+Tab to get back to the terminal and then hit ctrl+c to terminate).
This problem is caused by Electron not drawing anything in OpenGL. I resolved this issue by either downgraded Electron to version 1.4.5 or to upgrade to the latest beta (version 5.0.0). Go to your MagicMirror folder and run either of these two commands,

npm install electron@1.4.15
or
npm install electron@beta

I can't recall if I had to do a npm rebuild or a npm install afterwards but give it a try if the above does not resolve the issue.

dxfan227, 2019-04-09:
The answer above is correct. I tried to do the mouse pointer stuff and came back to a blank screen.  You need to add the desktop auto start info in the new file you create as it takes precede t over the system one

sdetweil, 2019-09-01:
my new install script now handles lots of different approaches to screen saver..
```
https://forum.magicmirror.builders/topic/10171/anyone-want-to-try-updated-installer
```
also have JUST the screensaver part for testing 
```
https://forum.magicmirror.builders/topic/11052/new-script-for-screensaver-disable
```
once tested more, I will submit  installer pr

AntumArk, 2019-09-01:
Somehow I could not login into desktop environment on my rpi3b+. Tried installing lxde and apparently it was missing (no idea how or when) now it is working alright.

closed 2019-11-07 by stale[bot]