MagicMirrorOrg/MagicMirror #1478: Bug in newsfeed module using `removeStartTags` on a description

michael5r, 2018-12-07:
**Platform**: Raspberry Pi 3, Mac
**Node Version**: 8.9.0
**MagicMirror Version**: 2.5

**Description**: 
There's a bug in the default `newsfeed` module in `newsfeed.js` when `removeStartTags` is used on the description - it's copying the sliced result back into the title, instead of the description.

Line 141
```js
this.newsItems[this.activeItem].title = ... [snip]
```
should be:

```js
this.newsItems[this.activeItem].description = ... [snip]
```

michael5r, 2018-12-10:
I'd be happy to open a PR for this tiny change, but I was unable to find code contribution guidelines.

MichMich, 2018-12-10:
PR to the develop branch is more than welcome. Thanks! Dont forget to add the changelog.

MichMich, 2018-12-19:
Merged!

closed 2018-12-19 by MichMich