MagicMirrorOrg/MagicMirror #1487: MagicMirror² unable to install dependencies

skyales, 2018-12-26:
When submitting a new issue, please supply the following information:

**Platform**: Raspbian Stretch with desktop and recommended software      Electron version (I don't know how to check that) *and* your hardware (Raspberry 3 B+).

**Node Version**:1.4.21

**MagicMirror Version**: I ran the newest command @ the MM Website.

**Description**: Hey guys, I want to build a MagicMirror for a school project, but the installation doesn't work properly.
I have installed the newest version of Raspbian and set up location etc.
During installation there are many errors with NPM and at least "unable to install dependencies".
Can somebody please help me? I know my Raspberry is in german, but the relevant codes are in english :)

Thank you & Merry Christmas

`pi@raspberrypi:/opt $ bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

$$\ $$\ $$\ $$\ $$\ $$\ $$$$$$
$$$\ $$$ | _| $$$\ $$$ |_| $$ $$
$$$$\ $$$$ | $$$$$$\ $$$$$$\ $$\ $$$$$$$\ $$$$\ $$$$ |$$\ $$$$$$\ $$$$$$\ $$$$$$\ $$$$$$\ _/ $$ |
$$$$$$ $$ | _$$\ $$ $$\ $$ |$$ |$$$$$$ $$ |$$ |$$ $$\ $$ $$\ $$ $$\ $$ $$\ $$$$$$ |
$$ $$$ $$ | $$$$$$$ |$$ / $$ |$$ |$$ / $$ $$$ $$ |$$ |$$ | _|$$ | _|$$ / $$ |$$ | _|$$ /
$$ |$ /$$ |$$ __$$ |$$ | $$ |$$ |$$ | $$ |$ /$$ |$$ |$$ | $$ | $$ | $$ |$$ | $$ |
$$ | _/ $$ |$$$$$$$ |$$$$$$$ |$$ |$$$$$$$\ $$ | _/ $$ |$$ |$$ | $$ | $$$$$$ |$$ | $$$$$$$$
_| _| _| _$$ |_| _|_| _|_|_| __| _/ _| _______|
$$\ $$ |
$$$$$$ |
_____/

Updating packages ...
OK:1 http://archive.raspberrypi.org/debian stretch InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Paketlisten werden gelesen... Fertig
Installing helper tools ...
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
build-essential ist schon die neueste Version (12.3).
curl ist schon die neueste Version (7.52.1-5+deb9u8).
git ist schon die neueste Version (1:2.11.0-3+deb9u4).
unzip ist schon die neueste Version (6.0-21).
wget ist schon die neueste Version (1.18-5+deb9u2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v5.1.0
Installed Node version: v8.11.1
No Node.js upgrade necessary.
Cloning MagicMirror ...
Klone nach 'MagicMirror' ...
remote: Enumerating objects: 282, done.
remote: Counting objects: 100% (282/282), done.
remote: Compressing objects: 100% (251/251), done.
remote: Total 282 (delta 34), reused 127 (delta 21), pack-reused 0
Empfange Objekte: 100% (282/282), 385.56 KiB | 375.00 KiB/s, Fertig.
Löse Unterschiede auf: 100% (34/34), Fertig.
Cloning MagicMirror Done!
Installing dependencies ...
(node:11272) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated express-ipfilter@0.3.1: This package is no longer being maintained and contains security vulnerabilities
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm ERR! Error: Method Not Allowed
npm ERR! at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR! at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR! at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR! at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this entire log,
npm ERR! including the npm and node versions, at:
npm ERR! http://github.com/npm/npm/issues

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.

electron-chromedriver@1.7.1 install /home/pi/MagicMirror/node_modules/spectron/node_modules/electron-chromedriver
node ./download-chromedriver.js

npm WARN deprecated circular-json@0.3.3: CircularJSON is in maintenance only, flatted is its successor.
npm WARN deprecated browserslist@2.11.3: Browserslist 2 could fail on reading Browserslist >3.0 config used in other tools.
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR! /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0
Unable to install dependencies!
`

skyales, 2018-12-27:
I don't know why, but I ran
sudo npm install -g npm
Then deleted the magicmirror folder and installed magicmirror again.

Now it works :D

closed 2018-12-30 by MichMich

Vaishnavikumara, 2019-07-10:
I am facing the same issue. Please help

Vaishnavikumara, 2019-07-10:
Hi @skyales ,I tried what you said , but it is still showing the same error